ptvoinfo/zigbee-configurable-firmware #122: CC2652P1 and PSM mode

axelhorn, 2022-05-27:
Many thanks PTVO for your firmware, which has already saved my projects several times. Now I have bought a license for this firmware to use PSM on E72 modules (CC2652P1) and only now I can see that PSM mode with this microcontroller is not working properly. Also, the "External Sensor Power Control" output does not work properly because the chip does not sleep. It worries me very much.
Is there any chance that in subsequent versions of this firmware the problem of a great CC2652P1 chip will be solved?

ptvoinfo, 2022-05-28:
Could you please show your firmware settings?

axelhorn, 2022-05-28:
Ofcourse.
[image]
[image]

axelhorn, 2022-05-29:
Can you at least confirm that the problem is in firmware or I do something wrong or I have a chip damaged.

ptvoinfo, 2022-05-29:
1. Wake up delay is empty.
2. How did you test that the "External Sensor Power Control" output does not work?
3. Probably, you should swap ADC and internal temperature (ADC should be first).
4. I don't know your schematic. Are you confident that "External Sensor Power Control" should be "Pull down"? The internal pull-up/down resistor is not recommended for PSM.

axelhorn, 2022-05-29:
1. I'll try without and with Wakeup delay = 1000ms.
2. GPIO10 drives mosfet transistor. But when I saw that were problems, I disconnected P10 from everything and measure it by oscilloscope. It allways stay high. Only when I press reset chip, port goes low for pressing time.
3. OK, I replaced now, thanks. But it looks like it doesn't matter in my problem.
4. Again - i check every "pull" in options. Port is allways high. I try "inverse" option, it makes port permanent low. Also I try external pull down option with external resistor 4k7 connected to mains. It's still the same.
5. Chip module coneumes about 0,76mA wen paired and idle. When LED blinks (probably then it transmits the data) it is about 8mA.

axelhorn, 2022-05-29:
6. I checked everything above on other port for sure. Same... :(

ptvoinfo, 2022-05-30:
Please try the following configuration. You should define the "External Sensor Power Control" pin immediately before a sensor or sensors that allow power control.
[image]

FIY: the internal temperature sensor or any empty channel resets the previously defined "External Sensor Power Control" pin.

The external pull-up resistor should be about 2 MOhm for smaller current leakage. The 4.7K resistor is useless because an internal pull-up/pull-down resistor is better.

axelhorn, 2022-05-30:
Yes, BINGO! 
"_the internal temperature sensor or any empty channel resets the previously defined "External Sensor Power Control" pin_"
is a game changer.  Big THANK YOU.
It may be time to update the instructions on your website. ;)
And another one question: Why do physical inputs (analog, temperature) set as outputs? Logic would order to do so on the inputs.

ptvoinfo, 2022-05-31:
1. I think it is described in the documentation https://ptvo.info/zigbee-configurable-firmware-features/power-saving-mode/
😄 What could I add?
2. Input and Output are important for GPIO only and some rare sensors. ADC on an output allows configuring additional options and the configuration becomes a better readable.

axelhorn, 2022-05-31:
OKAY. I read this text many times, but I did not pay attention to the record: "_This output should be defined immediately... or GPIO resets the “External sensor power control” setting_". Mea Culpa.
One more question - do I understand that "Wakeup Delay" turns on external sensors xxx ms **before** reading and sending data, so that they have time to collect data?

ptvoinfo, 2022-05-31:
> Wakeup Delay" turns on external sensors xxx ms before reading and sending data, so that they have time to collect data?

Yes, correct.

axelhorn, 2022-05-31:
Nice, eot ;)

closed 2022-05-31 by ptvoinfo