ptvoinfo/zigbee-configurable-firmware #123: Pulse switch doesn’t respect inversed on startup

vwir, 2022-06-04:
When I use pulse switch output in CC2530 then after power outage it always starts with low state, even in inversed mode. In inversed mode low state means a pulse, so after power outage it always starts with an infinite pulse. After first click the pulse clears and starts working correctly, but after following power outage I run into this problem again. I tried things like ‚remember state’ and so on, but none of it works.

ptvoinfo, 2022-06-06:
@vwir Thank you! I can confirm this bug. I'll fix in the next version.

ptvoinfo, 2022-06-06:
@vwir Please try the following update:
https://ptvo.info/download/ptvo-firmware-2022-06-06.zip

maciejblaz, 2022-06-06:
You are quick. I'll test it in the evening and will let you know.

ptvoinfo, 2022-06-10:
You are welcome!

closed 2022-06-10 by ptvoinfo