ptvoinfo/zigbee-configurable-firmware #124: Group switch - Interlock mode - Output switch appears closed but the action is not performed

mihsu81, 2022-06-04:
Hi @ptvoinfo,

Thanks a lot for the awesome job. 

I'm using a group switch in Interlock mode to make sure only 1 speed is selected on my kitchen hood (Output 3, 4 and 5).
While the interlock mode works fine when physical buttons are pressed, when I press the Endpoint switches in Z2MQTT, the previous switch is being switched off, but the inverted pulse is never sent, or at least that's what it looks like. If I press again the same Endpoint  switch in Z2MQTT, it gets turned off and the pulse is sent again actually switching physically to the selected speed.

Here's an example that will make it easier to understand:

Using physical buttons
1. Press Speed 1 physical button driving Input 3, Output 3 sends the inverted pulse switching the kitchen hood to Speed 1, in Z2MQTT the associated I3 switch turns on, the kitchen hood switches on Speed 1.
2. Press Speed 2 physical button driving Input 4, Output 4 sends the inverted pulse switching the kitchen hood to Speed 2, in Z2MQTT the associated I4 switch turns on and I3 turns off, the kitchen hood switches to Speed 2.

Using virtual buttons
1. Turn on the Endpoint I3 switch in Z2MQTT, the kitchen hood switches to Speed 1.
2. Turn on the Endpoint I4 switch in Z2MQTT, the Endpoint I3 switch turns off, the kitchen hood remains set to Speed 1 and never switches to Speed 2.
4. Turn off the Endpoint I4 switch in Z2MQTT, the kitchen hood switches to Speed 2.

I've tried shorter and longer impulses, but it didn't help.
Is the pulse never sent, or 2 pulses are sent at he seme time and the kitchen hood can't deal with 2 button presses?

Thanks in advance for your help solving this issue.

[image]
[image]

ptvoinfo, 2022-06-06:
Hi @mihsu81 

I've checked this. The firmware should send two pulses when you change an output state in Z2M.  The first pulse switches off the active output, and the second pulse switches on the new active output.  The firmware generates both pulses at the same time.

mihsu81, 2022-06-06:
Hi @ptvoinfo,

I did some more troubleshooting and realised that indeed the board of the kitchen hood can't handle 2 pulse switches at the exact same time. If there's a difference of at least 100ms, both pulse switches are interpreted.
Would it be possible to have a configurable delay between the pulse switch which turns off the active outputs and the pulse switch which turns on the inactive output?
Or even better, a type of grouping which assumes that turning on 1 output automatically turns off the other outputs included in the group without having to actually send a signal?

Thanks in advance.

ptvoinfo, 2022-06-06:
Probably, you can try the "Bistable relay" output type instead of "Pulse switch", but define "Bistable relay off" on any virtual pin.

mihsu81, 2022-06-06:
Hi @ptvoinfo,
Thanks a lot for the suggestion. I've tried it and works fine when switching from one speed to the other, but it breaks normal On/Off functionality because it requires me to press a button (physical or virtual) 2 times before turning the hood off.

[image]

Changing the 3 Inputs to Switch / Toggle helped with the usage of physical buttons because each press of a button physically turned On or Off the Output, but in Z2MQTT the Buttons always appear Off. Also, I have to press a virtual button twice before a pulse is sent.

[image]

ptvoinfo, 2022-06-07:
Sorry, but I do not have other suggestions. Unfortunately, I cannot add the necessary changes to the firmware.

mihsu81, 2022-06-07:
Hi @ptvoinfo, 
Even if I purchase a premium version? Or that still wouldn't compensate the additional development work?

ptvoinfo, 2022-06-07:
Yes, the firmware cost does cannot not cover the cost of additional work.

mihsu81, 2022-06-07:
Ok. Thank you anyway for your help and for the awesome job with this software. 👍

closed 2022-06-07 by mihsu81

mihsu81, 2022-06-07:
Thinking about it, would it be possible to implement a delay of 50-100ms between switching off one output and switching on the other? Physically I can see it working with a short delay between the pulses.

ptvoinfo, 2022-06-08:
Maybe, you can add a capacitor with a resistor on an output that will add a delay?