ptvoinfo/zigbee-configurable-firmware #125: CC2530 E18-MSB1-PCB and HC-SR04

Tb7386, 2022-06-15:
Hello,
I flash E18-MS1-PCB with PTVO firmware with same configuration for connect HC-SR04.
[image]
Status led slow flash when it’s associate with zigbee2mqtt but I don’t receive any distance value.
I check signal with SALEAE on pin P05(18) and pin P06(17) :
– P05 stay at low level.
– P06 stay at low level.
Firmware never send trigger signal.
[image]

What I do false ?

ptvoinfo, 2022-06-16:
Hello,

Did you enable periodic reports on the "Expert" tab?

Tb7386, 2022-06-16:
Hello, thank you for your help.
My sensor works with Arduino and an example sketch like: https://www.carnetdumaker.net/articles/mesurer-une-distance-avec-un-capteur-ultrason-hc-sr04-et-une-carte-arduino- genuine/
I'm testing this configuration:
[image]

Analyzing it with Saleae shows the trigger signal and the echo return signal with correct measure (~ 1.5m)
[image]

But no data sent to Zigbee2MQTT.
I tried to modify Manufacture and model ID. I import converter for Z2M.
[image]

 In Z2M, device seen like ptvo.switch. No distance measurement in log.

Tb7386, 2022-06-16:
I tried to change PIN too. But issue is the same

closed 2023-02-11 by ptvoinfo

1GOR19, 2023-12-18:
> Здравствуйте, спасибо за помощь. Мой датчик работает с Arduino, и пример эскиза такой: [https://www.carnetdumaker.net/articles/mesurer-une-distance-avec-un-capteur-ultrason-hc-sr04-et-une-carte-arduino-подлинное](https://www.carnetdumaker.net/articles/mesurer-une-distance-avec-un-capteur-ultrason-hc-sr04-et-une-carte-arduino-) / Я тестирую эту конфигурацию: [image]
> 
> Анализ с помощью Saleae показывает сигнал запуска и обратный эхо-сигнал с правильным размером (~ 1,5 м). [image]
> 
> Но данные в Zigbee2MQTT не отправляются. Я попытался изменить идентификатор производителя и модели. Импортирую конвертер для Z2M. [image]
> 
> В Z2M устройство отображается как ptvo.switch. В журнале нет измерения расстояния.

Were you able to transfer distance data to Z2M?