ptvoinfo/zigbee-configurable-firmware #126: shtc3 не работает

Garkus98, 2022-06-25:
shtc3 на последней прошивке от 30 мая не работает на 2652p и 2530.

ptvoinfo, 2022-06-29:
I cannot confirm this problem. Tested on CC2652P. The I2C address is 0x70 (hex) or 112 (dec)

Garkus98, 2022-06-29:
Я прошел логическим анализатором и там тишина. на 2652p

ptvoinfo, 2022-06-29:
Please attach a screenshot with your firmware settings.

Garkus98, 2022-06-29:
[image]
[image]

ptvoinfo, 2022-06-29:
Do you really have the external pull-up resistor on I2C wires? The standard SHTC3 sensor does not have them inside.

Garkus98, 2022-06-29:
да там есть

ptvoinfo, 2022-07-02:
The firmware checks the ID of the SHTC3 sensor. Some Chinese clones use another sensor ID. Therefore, the firmware cannot detect these sensors properly.

Garkus98, 2022-07-02:
Да Мы с вами уже это увидели когда я вам отправил с логического анализатора данные и вы отправили спец версию прошивки.

closed 2022-07-25 by ptvoinfo