ptvoinfo/zigbee-configurable-firmware #133: Read 7 Segment Display

Danit2, 2022-07-03:
Is it possivke to read out a 7 segment display with your firmware? 
There is everywhere a lot of 7 segment displays and it would be great when we can bring them in Z2M.

ptvoinfo, 2022-07-04:
Sorry, but the firmware does not support these displays.

bojanpotocnik, 2022-07-05:
@Danit2 you can connect anodes of relevant 7-segment display LEDs to inputs on your module and then process them in the e.g. Home Assistant or wherever you're sending your data.
I have done this with my garage doors, to be able to (for safety, in addition to being also connected directly to the encoder and end switches) always check the garage door state as perceived by the original control system (which displays the state on 7-segment display).
[image]
[image]

Danit2, 2022-07-05:
@bojanpotocnik Thanks for your tip. But i want to read a 3 digit 7 segment display from my dryer. 
This work with a multiplexer. So it is not so easy to read this. 
I dont know how to bring this info in my smarthome.

bojanpotocnik, 2022-07-05:
If I was you, I would do it using ESP32 Cam module from AliExpress and ESPHome.
Or even easier for a drier: just measure the power consumptions - when it rises above the trigger value for at least e.g. 5 minutes and then falls under the trigger value (standby) for X minutes, it has finished - that's how I did it.

closed 2022-07-21 by Danit2