ptvoinfo/zigbee-configurable-firmware #134: Configuration for Sonoff Zigbee 3.0 Usb Dongle Plus

jairbj, 2022-07-18:
Hi friends, can anyone provide configurations for router firmware on SONOFF Zigbee 3.0 USB Dongle Plus?

ptvoinfo, 2022-07-18:
You can find the configuration in the "Presets" folder and load it in the "File" menu.

jairbj, 2022-07-19:
> You can find the configuration in the "Presets" folder and load it in the "File" menu.

Thank you a lot. I didn't see there is a "Sonoff Dongle Pluss" preset, I Just saw the other files.

ptvoinfo, 2022-07-19:
Hmm. I've checked the latest version. The "sonoff_3_usb_dongle_plus.ini" file is there.

jairbj, 2022-07-19:
> Hmm. I've checked the latest version. The "sonoff_3_usb_dongle_plus.ini" file is there.

Yes it's there. It was my fault. 

Thank you a lot

closed 2022-07-19 by jairbj

jairbj, 2022-10-11:
Yes it's there, it was my fault.

I just didn't see at the first time.

Thank you.