ptvoinfo/zigbee-configurable-firmware #138: Sonoff serial bootloader (SBL) toggle

janmashat, 2022-08-02:
Hi, 2 weeks ago I succesfully flashed [cc2652p.hex](https://github.com/ptvoinfo/zigbee-configurable-firmware/tree/master/bin) by running:

```
docker run \
--device /dev/serial/by-id/usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_bee4...6eb0-if00-port0:/dev/ttyUSB1 \
-e FIRMWARE_URL=https://example.com/cc2652p.hex \
ckware/ti-cc-tool \
-ewv \
-p /dev/ttyUSB1 \
--bootloader-sonoff-usb
```

Now I tried to upgrade to the new 2022-07-21 version by running the same command, however it failed with:

```
Downloading firmware from https://example.com/cc2652p.hex
Firmware file:   'cc2652p.hex'
sonoff
Opening port /dev/ttyUSB1, baud 500000
Reading data from cc2652p.hex
Your firmware looks like an Intel Hex file
Connecting to target...
ERROR: Timeout waiting for ACK/NACK after 'Synch (0x55 0x55)'
```

Is it possible that the config in cc2652p.hex isn't compatible with [`--bootloader-sonoff-usb`](https://github.com/JelmerT/cc2538-bsl/pull/114)?

If that's the case, can it be added? And in the meantime I assume I'll need to open the case and hold some buttons to invoke SBL?

ptvoinfo, 2022-08-02:
Hi,

Unfortunately, it is a mistake to use cc2652p.hex without configuration. It is a template only. SBL is not active there. Now, you need a XDS110 or similar JTAG flasher to upload a new firmware.

janmashat, 2022-08-02:
Oh no 😞 even the cheapest c/JTAG flasher (CC-DEVPACK-DEBUG) is almost twice as expensive as the dongle itself - so I would rather get a couple more zigbee dongles. It seems I'm not the only one to make this [mistake](https://github.com/JelmerT/cc2538-bsl/issues/113#issuecomment-1003533102) so a disclaimer in the [README](https://github.com/ptvoinfo/zigbee-configurable-firmware/blob/master/README.md) might help future visitors 😉

janmashat, 2022-08-02:
Actually I found a bunch more people making the same mistake [here](https://github.com/Koenkk/Z-Stack-firmware/issues/316) - wouldn't it make sense to enable SBL in the template hex?

ptvoinfo, 2022-08-03:
It is useless because the SBL pin should be configured too, and it may change for different boards.

janmashat, 2022-08-03:
Isn't DIO_15 the default BSL pin for CC2652P? At least that's what it seems to be according to [this table](https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_3.x.0/bin).

ptvoinfo, 2022-08-03:
Currently, SBL isn't activated at all in the template HEX file. Sorry.

janmashat, 2022-08-03:
I understand - I'm suggesting to activate it in order to make it easier for others in the future.

ptvoinfo, 2022-08-05:
@janmashat Probably good new for you. SBL is enabled in the firmware, but the SBL function does not use any pins to active it. I know, there is a method to start flashing a firmware without pulling a pin. But I didn't try it myself yet.

closed 2022-12-07 by ptvoinfo