ptvoinfo/zigbee-configurable-firmware #143: How to use commandToggle?

vanagnostos, 2022-09-05:
Hello. What is the trick to use the new feature added here: https://github.com/ptvoinfo/zigbee-configurable-firmware/issues/104 ?
Whatever I try it keeps sending ON/OFF. I am using the latest version and PSM. Goal is to send "toggle" using a push button. Thanks!

ptvoinfo, 2022-09-06:
Hi! Could you please attach a screenshot with your configuration? I'll test it.

vanagnostos, 2022-09-06:
Hi. I first tried same setup as here https://github.com/ptvoinfo/zigbee-configurable-firmware/issues/104#issuecomment-1089201249 and then at least 5-6 other combinations, here are the last two I tried:  

Board type: CC2530
Device type: PSM
Manufacturer ID: 64448
Manufacturer name: vaskoa
Model ID: vaskoa
Update firmware's timestamp : 2022-09-06
Wait for commands after a report in PSM: Yes
Disable configuring the interval remotely in run-time: Yes
Set default reporting interval (s): 0

Output pins:
P37: Output 2, GPIO (Role: Generic), Remember state

Input pins:
P02: Input 1, GPIO, External pull-down, Switch (Bind command: Toggle)
P12: Input 2, GPIO, External pull-down, Link to out 2 (Bind command: Toggle)

I am using 1mo pull down resistor and a button. Before each attempt I export and replace the custom converter.

ptvoinfo, 2022-09-06:
I've checked this setup. The firmware sends the Toggle command in both cases. Do you have a sniffer to capture zigbee packet on your side?

vanagnostos, 2022-09-06:
Maybe I am doing something wrong, not sure. Tried to check the documentation but this functionality is not mentioned.  

I don't have a sniffer but I have zigbee2mqtt debug log. First thing that I notice is that there is no "toggle" mentioned in exposed actions. 

Here is the log: 

```
2022-09-06 16:47:12Successfully removed device '0x00124b0014db0079' (block: false, force: false)
info 2022-09-06 16:47:12MQTT publish: topic 'zigbee2mqtt/bridge/response/device/remove', payload '{"data":{"block":false,"force":false,"id":"0x00124b0014db0079"},"status":"ok","transaction":"y9dt0-9"}'
info 2022-09-06 16:48:11MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"commit":"e0a9323d","coordinator":{"meta":{"maintrel":3,"majorrel":2,"minorrel":6,"product":0,"revision":20190619,"transportrev":2},"type":"zStack12"},"log_level":"debug","network":{"channel":11,"extendedPanID":"0xdddddddddddddddd","panID":6754},"permit_join":true,"version":"1.27.0"}'
info 2022-09-06 16:50:56Device '0x00124b0014db0079' joined
info 2022-09-06 16:50:56MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x00124b0014db0079","ieee_address":"0x00124b0014db0079"},"type":"device_joined"}'
info 2022-09-06 16:50:56MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0x00124b0014db0079"},"type":"device_connected"}'
info 2022-09-06 16:50:56Starting interview of '0x00124b0014db0079'
info 2022-09-06 16:50:56MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x00124b0014db0079","ieee_address":"0x00124b0014db0079","status":"started"},"type":"device_interview"}'
info 2022-09-06 16:50:56MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0x00124b0014db0079"},"type":"pairing"}'
debug 2022-09-06 16:50:58Device '0x00124b0014db0079' announced itself
info 2022-09-06 16:50:58MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x00124b0014db0079","ieee_address":"0x00124b0014db0079"},"type":"device_announce"}'
info 2022-09-06 16:50:58MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0x00124b0014db0079"},"type":"device_announced"}'
debug 2022-09-06 16:50:59Received Zigbee message from '0x00124b0014db0079', type 'readResponse', cluster 'genBasic', data '{"modelId":"great.vasko"}' from endpoint 1 with groupID 0
debug 2022-09-06 16:50:59Skipping message, definition is undefined and still interviewing
debug 2022-09-06 16:50:59Received Zigbee message from '0x00124b0014db0079', type 'readResponse', cluster 'genBasic', data '{"manufacturerName":"ptvo.info"}' from endpoint 1 with groupID 0
debug 2022-09-06 16:50:59Skipping message, definition is undefined and still interviewing
debug 2022-09-06 16:50:59Received Zigbee message from '0x00124b0014db0079', type 'readResponse', cluster 'genBasic', data '{"powerSource":3}' from endpoint 1 with groupID 0
debug 2022-09-06 16:50:59Skipping message, definition is undefined and still interviewing
debug 2022-09-06 16:50:59Received Zigbee message from '0x00124b0014db0079', type 'readResponse', cluster 'genBasic', data '{"zclVersion":1}' from endpoint 1 with groupID 0
debug 2022-09-06 16:50:59Skipping message, definition is undefined and still interviewing
debug 2022-09-06 16:50:59Received Zigbee message from '0x00124b0014db0079', type 'readResponse', cluster 'genBasic', data '{"appVersion":1}' from endpoint 1 with groupID 0
debug 2022-09-06 16:50:59Skipping message, definition is undefined and still interviewing
debug 2022-09-06 16:51:00Received Zigbee message from '0x00124b0014db0079', type 'readResponse', cluster 'genBasic', data '{"stackVersion":1}' from endpoint 1 with groupID 0
debug 2022-09-06 16:51:00Skipping message, definition is undefined and still interviewing
debug 2022-09-06 16:51:00Received Zigbee message from '0x00124b0014db0079', type 'readResponse', cluster 'genBasic', data '{"hwVersion":1}' from endpoint 1 with groupID 0
debug 2022-09-06 16:51:00Skipping message, definition is undefined and still interviewing
debug 2022-09-06 16:51:00Received Zigbee message from '0x00124b0014db0079', type 'readResponse', cluster 'genBasic', data '{"dateCode":"2022-09-06"}' from endpoint 1 with groupID 0
debug 2022-09-06 16:51:00Skipping message, definition is undefined and still interviewing
debug 2022-09-06 16:51:00Received Zigbee message from '0x00124b0014db0079', type 'readResponse', cluster 'genBasic', data '{"swBuildId":"2022-09-06"}' from endpoint 1 with groupID 0
debug 2022-09-06 16:51:00Skipping message, definition is undefined and still interviewing
info 2022-09-06 16:51:00Successfully interviewed '0x00124b0014db0079', device has successfully been paired
info 2022-09-06 16:51:00Device '0x00124b0014db0079' is supported, identified as: Custom devices (DiY) [Configurable firmware](https://ptvo.info/zigbee-configurable-firmware-features/) (great.vasko)
info 2022-09-06 16:51:00MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"[Configurable firmware](https://ptvo.info/zigbee-configurable-firmware-features/)","exposes":[{"access":1,"description":"Remaining battery in %","name":"battery","property":"battery","type":"numeric","unit":"%","value_max":100,"value_min":0},{"access":1,"description":"Triggered action (e.g. a button click)","name":"action","property":"action","type":"enum","values":["single","double","triple","hold","release"]},{"acces [... cut ...]
2022-09-06 17:22:06Received Zigbee message from '0x00124b0014db0079', type 'attributeReport', cluster 'genMultistateInput', data '{"presentValue":1}' from endpoint 2 with groupID 0
info 2022-09-06 17:22:06MQTT publish: topic 'zigbee2mqtt/0x00124b0014db0079', payload '{"action":"single_l2","click":"l2_single","linkquality":92}'
debug 2022-09-06 17:22:06Received Zigbee message from '0x00124b0014db0079', type 'attributeReport', cluster 'genMultistateInput', data '{"presentValue":1}' from endpoint 2 with groupID 0
info 2022-09-06 17:22:06MQTT publish: topic 'zigbee2mqtt/0x00124b0014db0079', payload '{"action":"single_l2","click":"l2_single","linkquality":94}'
```

ptvoinfo, 2022-09-07:
The firmware sends the "toggle" command to a bound device. It does not appear in Z2M.

vanagnostos, 2022-09-09:
Okay, I see. I kinda expected something like this (this is what other devices I have do): 

`debug 2022-09-09 10:08:35 Received Zigbee message from 'Led Remote', type 'commandToggle', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 904`

closed 2022-09-09 by vanagnostos