ptvoinfo/zigbee-configurable-firmware #151: Model ID is always CC2530.ROUTER regardless from settings

flaretom, 2022-10-30:
Hi,
I tried to configure a device with a Model ID (following https://hackaday.io/project/168959-let-me-control-you-hitachi-air-conditioner), but my Zigbee2MQTT is always showing CC2530.ROUTER instead of the programmed RAS-LJ22Y_ADPT. I already deleted the device in Zigbee2MQTT  and forced a repairing, but to no avail.
Is this a feature only for the Premium version (but from the Wiki it doesn't look like)?

BR Tom

ptvoinfo, 2022-10-30:
Hi Tom,

1. Delete your device in Z2M.
2. Restart Z2M
3. Join your device again.

flaretom, 2022-11-03:
Hello ,
Sorry I was only today able to test your hint. The restart of Zigbee2MQTT was the important part.
And now it worked as expected :). 

Thank You very much for Your kind support.

closed 2022-11-03 by flaretom