ptvoinfo/zigbee-configurable-firmware #152: Feature request: Single button dimmer

geelenbert, 2022-10-31:
Hi There,

I have been working with the firmware for a long time now and i really love it.

But there is one feature that would create the perfect single button light dimmer. I can easily convert my in wall switched to momentary switched by adding a small spring to them, but a separated up and down button would be way more difficult and requires a lot of new hardware.

The function is already almost in there, but instead of 2 separated buttons (Level up and level down) , i would need it to be in one single.

Function description:

1 momentary push button would be required.
- On single click (<350ms) -> Toggle light
- On long push (> 350ms) -> Move level up with On -> On release -> Stop move
- On 2nd Long Push (> 350ms) -> Move level down with off -> On release -> Stop move
- On 3rd Long Push (> 350ms) -> Move level up with On -> On release -> Stop move
- .... etc....
Bonus would be to expose a double click event, to trigger some scenes.

For me it would be essential that the light controls can be binded directly to a group, so it could work without a coordinator.

ptvoinfo, 2023-02-08:
@geelenbert Hi! Could you please test the latest version where I've added this feature?

geelenbert, 2023-02-08:
@ptvoinfo How would i configure this ?

ptvoinfo, 2023-02-08:
@geelenbert Hmm. It is a relay module. It cannot move level up or down. You need a dimmer.

geelenbert, 2023-02-08:
@ptvoinfo Yes i understand, but i don't want to control the local relay, I just want the input to control the Zigbee light groups

ptvoinfo, 2023-02-08:
Sorry, but I didn't understand your initial request properly. I thought, you mean local control of a dimmable output.

geelenbert, 2023-02-08:
Ok. Could you give me an configuration example that should demonstrate the function so i can test and experiment with it ?

ptvoinfo, 2023-02-08:
You cannot test it because you do not have a corresponding device. This update does not allow sending these commands to a remote device.

closed 2023-05-06 by ptvoinfo