ptvoinfo/zigbee-configurable-firmware #153: How to set Motion Sensor Occupancy Timeout

maxmib, 2022-11-01:
My motion sensor sends the off state immediately after triggered in ptvo fiemware ,but in fact it  detects again in 5 seonds after triggered , so  is there a way to increase the  occupancy timeout in ptvo firmware? Thanks.

ptvoinfo, 2022-11-04:
Sorry, but you cannot do it in the firmware. Usually, motion sensors have a possibility to configure the occupancy timeout inside using a resistor or jumpers.

maxmib, 2022-11-06:
> Sorry, but you cannot do it in the firmware. Usually, motion sensors have a possibility to configure the occupancy timeout inside using a resistor or jumpers.

Thanks,my motion sensor's occupancy can't be configured in physical, maybe I can configure this in zigbee2mqtt or homeassistant.And another question about input switch in psm mode,only one switch should be set on a port,so how about  external wake-up gpio  ,should it be treated as a switch and can't share one port with other  switch ? for example ,I set input1 as external wake-up using p12 and input2 as a switch  using p15, is this setting ok ?

closed 2023-02-11 by ptvoinfo