ptvoinfo/zigbee-configurable-firmware #156: Troubles binding groups

geelenbert, 2022-11-07:
I'm having troubles with binding a group to a switch.

My hardware is the SonOff ZBMini device.

Loading the provided example works as expected:
[image]

When using the Switch input (P08/Input2) the relay latches. 
Using Z2MQTT i can bind a group to endpint 2 on/off. When the relay is switched onor off, the light group is also turned on and off.

What i want to achieve is:
Directly bind groups to a button and use the toggle command to toggle the binded group, without the local relay. 
The relay should be controlled externaly trough zigbee. 
[image]

Using this configuration, i'm unable to bind a group to "input 3". 

is there something wrong with my configuration, of should this work as discribed ?

geelenbert, 2022-11-21:
I don't want to be rude, but is there anyone who can help me to understand if it is possible what i would like to achieve  ?

ptvoinfo, 2022-12-07:
@geelenbert 
I'm sorry about the delay with my answer. Binding to a group works as expected. I've tested it with Zigbee2MQTT.

geelenbert, 2022-12-07:
@ptvoinfo no problem, thanks for the reply.

But can you maybee explain the expected behaviour of the Bind command function?
Should this work on any input independend of the output?

From my testing it appears that binding only works on outputs ?

Would you be able to confirm the expected behavior of the following configuration:
[image]

Input 1 would be used for configuring the device (factory reset etc...)

Output 2 would be an independent controllable output trough zigbee, not effected by any local buttons.

Input 4 would be able to bind to a group (using z2mqtt). 
When clicking the button, a toggle command would be send to the binded group. 
Long push, Double and tripple clicking would not send a toggle command, but would send the event to the multi_input cluster. 

Additional, this is a configuration for a Sonoff ZBMINI

ptvoinfo, 2022-12-07:
There are two parties in binding. One party sends a command (Input) and another party receives a command (Output).

1. Add some outputs to a group.
2. Bind an input to this group.

In this case, the input will send a command to a coordinator, that will distribute this command to all receivers (group members).

A device that sends commands may not belong to a group. 

But in your case, you may send a command from Input 4, and at the same time add Output 2 to a group that receives commands from Input 2.

I hope, I've described it clearly :).

geelenbert, 2022-12-07:
@ptvoinfo Using the configuration above results in these clusters:
[image]

[image]

Cluster 4 does not give the option to bind to anything. 

Is this a configuration error on my side, or is there a bug somewhere ?

povlhp, 2024-01-12:
We need genOnOff in an output cluster. How is this done ? This is needed to bind switch to light to run independent of z2m.