ptvoinfo/zigbee-configurable-firmware #157: Can't remotely change number of  LEDs (CC2530+CC2592)

DJTerentjev, 2022-11-13:
Please see attached configuration file. Can't remotely change the number of LEDs. I tried changing in the web interface of Z2M (Home Assistant) and send command to the topic: zigbee2mqtt/0x00124b001ec71cb2/l5/set
[image]
[z_led.txt](https://github.com/ptvoinfo/zigbee-configurable-firmware/files/9998146/z_led.txt)

ptvoinfo, 2022-11-14:
Did you reboot your device after the configuration change?

DJTerentjev, 2022-11-14:
No. I didn't. You are correct. If number of LED was changed, device have to be reset or power off and on. I was expected to change on the 'fly' 👍 .  Could you please mention it on your site in the remote configuration section. Also please correct information on the site. Right topic is zigbee2mqtt/[friedly_name]/set/[channel]. Thanks.

closed 2022-12-07 by ptvoinfo