ptvoinfo/zigbee-configurable-firmware #159: Constant 3.3V on pin 10 in PSM mode with External sensor power control

Belaial, 2022-11-18:
Hi,

The issue I am facing is that I built a Zigbee sensor with a CC2530 following this thread ( https://community.home-assistant.io/t/diy-zigbee-4x-ds18b20-temperature-measurement/246584 ), besides that thread I followed advice on ptvo.info to make it a PSM device (so I got a license) and I also used the feature "External sensor power control" on pin 10 to make the sensors only use power once needed (I use 5x DS18B20), all works fine so far but I was a bit surprised my sensor "only" lasted around 8-9 days on a 18650 battery charged to 3.6V (unsure how much power is stored in the battery at that charge but it's a high end Panasonic NCR18650B cell rated at 3400 mAh according to Google).

I took my multi-meter and hooked it up to GND and pin 10 to see if the "External sensor power control" was doing it's job and this is where my question comes, it says here ( https://ptvo.info/zigbee-configurable-firmware-features/power-saving-mode/ ) that it should go to HIGH state before reading the sensor value, but with my multi-meter I read a constant 3.3V power on pin 10, is this expected when using "External sensor power control"? I suspect this is what is draining my battery faster than expected, I thought I would see ~0V most of the time and 3.3V a short time while reading the sensors values, I also have the "Default reporting interval" set to 60.

I also measured the current draw on the VCC pin

No sensors attached was around 1.2 uA

With sensors attached was around 193.5 uA (and a spike for a few seconds somewhere around 8-17 mA when sending data, hard to say exact value just using a regular multi meter).

I have re-flashed the CC2530 and done all I can think of but it behaves the same way whatever I try, I am no expert in drawing electrical diagrams but here is my setup, hope you can understand it, I left out the pins I am not using.

[image]

Besides this diagram I assigned pin 11 to "source power" so I could graph that as well i Home Assistant. Without PSM mode my sensor lasted less than a day so I at least know the PSM mode did a great job!

Here is my PTVO config

[image]

[image]

ptvoinfo, 2022-11-19:
Hi Jonas,

Note 1: you should define the "External sensor power control" output **before** other sensors.
https://ptvo.info/zigbee-configurable-firmware-features/power-saving-mode/

Note 2: Probably, you can get better results with the internal pull-up resistor because the firmware disables it automatically when the device transits to the sleep state. Or you can increase the nominal of external pull-up resistor to 47K.

Belaial, 2022-11-19:
Hi,

Note 1: I completely missed that part, will redo my config and re-flash.

Note 2: Ok so I just remove the 4,7K resistor from my board and tell PTVO to use a internal pull up resistor instead? I guess I do the same for P10 where I have "External sensor power control" defined?
(no clue how that works, not that into electronics design, I just followed the guide from HA forums)

Note 3: I will increase it just to try and decide on a level I am happy with.

Note 4: Yes, 4.2V is a full charge, issue is that I did not have a better power source when building this, I just last night got these ( https://www.ebay.com/itm/143418755782 ) in the mail so will add one of those to the board, with that I can go full charge, before that 3.6V was the maximum safe limit for the C2530, or at least that is what I could find online.

Note 5: Point taken, I assume this does not consume that much extra energy having enabled? But I can also just remove it.

Belaial, 2022-11-19:
Hi again,

I have now re-flashed the CC2530 with the "External sensor power control" first in line and the sensors following afterwards, I also removed the "Source voltage" and put the reporting to 120 instead (will see how that works for my graphs, I use this for monitoring a wood burning boiler and a hot water tank connected to the boiler so when I heat it up I want quite "rapid" updates, afterwards it's fine if it reports slower but I will see what value works for me). I guess there is not good way of solving that? I mean having different report intervals depending if I am heating the system up or if it's just idle and I want to know how fast the temperatures are dropping and when it's time to start burning wood again.

I have yet to remove the 4,7K resistor and re-flash with "pull up", hopefully I understood it correctly but will patiently wait for a reply on Note 2: :)

I also after all this measured voltage on pin 10 and now it's 0V and goes to 3.3V just to read the sensors every 120 seconds! So it was just my mistake having the things in the wrong order, sorry for that mistake and thank you for noticing it and helping out!

closed 2022-12-07 by ptvoinfo