ptvoinfo/zigbee-configurable-firmware #168: LED firmware. Controlling Colour Temperature.

DJTerentjev, 2022-12-22:
Thank you for your effort to crate and constantly develop this firmware. It's probably easies way to create DIY ZigBee devises . Could you please add control for light temperature for RGB+CCT light with two PWM outputs.

[image]

ptvoinfo, 2022-12-23:
@DJTerentjev Sorry, but I don't have the corresponding device for testings and don't know how it should work.

DJTerentjev, 2023-01-17:
This option was added to configurator. Thank you!

closed 2023-01-17 by DJTerentjev

24021438, 2023-03-30:
I want to make a dual color temperature lamp, but I don't know how to set the color temperature channels L1L2, e.light_ brightness_ Colortemp (). withEndpoint ('l2 '), including brightness control. How do I fill in the channel control of two L1L2 channels? e.light_ brightness().withEndpoint('l2')

ptvoinfo, 2023-03-31:
@24021438 Please load the "gledopto" configuration file from the "Presets" folder.