ptvoinfo/zigbee-configurable-firmware #170: UART pins

wspeth, 2023-01-06:
I have failed so far to figure out the correct pins and there might be something wrong in the doku. Could you kindly check and clarify?

my device is a z1mini which carries an Ebyte E18 module and that has a TI CC2530 on board.
https://www.ti.com/lit/ug/swru191f/swru191f.pdf (CC253x User's Guide) shows UART RX on P0.2, TX (pin 17) on P0,3 (pin 16). Apparently Ebyte e18 connects those to pins 21 and 20,
highest port of CC2530 in P2.4

https://ptvo.info/zigbee-configurable-firmware-features/uart/ says "You should enable UART on P02 on an output pin in the configuration". When P02 is RX, shouldn't we expect it to be an input?

The dropdown control of your firmware configuration tool offers choices P00 to P37,  This cannot possibly refer to the ports P0.0 to P2.3. It seems to refer to the pins. On the other hand cc2530 has   pins 1 to 40 , yet no 0 on the chip and no 38 in the dropdown and the pins P2 and P3 on GND.

What is my mistake here? Is it really ports then like P0.0 to P2.4 in the dropdown list and all others I should not use?
Then only 13 options of the 8 inputs plus 8 outputs the tool offers can be used.

ptvoinfo, 2023-01-06:
1. "Input" and "Output" terms are applicable for GPIO only. If you configure a sensor or UART, you can treat "Output" as "Channel" or "Endpoint". 
2. P00 ... P23 are real hardware pins related to P0.0..P2.3. P30..P37 are virtual pins that you may use for some virtual GPIO inputs and outputs. https://ptvo.info/zigbee-configurable-firmware-features/gpio/
Please note, P00..P23 are logical pin names. You should check documentation for E18 to find a relation between hardware pin numbers and logical names.

wspeth, 2023-01-14:
> 1. "Input" and "Output" terms are applicable for GPIO only. If you configure a sensor or UART, you can treat "Output" as "Channel" or "Endpoint".
> 2. P00 ... P23 are real hardware pins related to [...]

got it! thanks for detailing. Helped me to set it up and it works 👍

closed 2023-01-15 by ptvoinfo