ptvoinfo/zigbee-configurable-firmware #171: SONOFF SNZB-01 not working!!!

bummer52, 2023-01-14:
Hi.
I have a SONOFF SNZB-01 button. I bought the premium version of the program. I made a firmware for him. When in the Device type is select (End device vithout route), everything works as it should, when Device type is select (power saving mode), it does not connect to the zigbee network at all.
P.S. In my premium version of programm is 8 inputs and outputs not a 16. its normally?

ptvoinfo, 2023-01-15:
Hi,

1. Could you please show me your settings? Did you try to place this button closely to the coordinator? Please note, the firmware tries to connect about 1 minute. You need to press a button periodically, to extend this interval.
2. The PSM firmware supports 8 channels only. There is the separate "16-channel" version.

bummer52, 2023-01-15:
[image]
[image]
Button placed from coordinator aprox 50cm. Why button with end device setting conect to coordinator ~10sec

ptvoinfo, 2023-01-15:
Could you please try to select the "CC2530 (without external oscillator)" board type?

bummer52, 2023-01-26:
I combine ptvo firmware with efekta converter and that work fine. this option work "CC2530 (without external oscillator)"

closed 2023-01-26 by bummer52