ptvoinfo/zigbee-configurable-firmware #173: Scale factors for HLW8032 don't do anything

axelhorn, 2023-01-18:
I connected HLW8032 in isolated version for tests to CC2530_CC2591 (output 2; P02 in firmware). The current in Z2M shows quite correctly. But the voltage is crazy – 420V instead of around 230V. When I swap L and N – it shows about 14V. As a result, the other parameters do not match.
The problem is that the firmware does not react at all to changing the “Scale factor voltage” and “Scale factor current” values. The only thing the firmware reacts to is changing from isolated to non-isolated, but then the readings are even weirder.
Am I doing something wrong or is there an error in the HLW8032 integration?

ptvoinfo, 2023-01-27:
@axelhorn I've verified it, and both scale factors work correctly.

Could you please show me a screenshot with your settings?

axelhorn, 2023-01-27:
Sorry ptvo and thanks for your answer.  
I tried on the new module and it was as I wrote. I bought a second new module and it is fine on it. Looks like the first one was factory damaged.