ptvoinfo/zigbee-configurable-firmware #174: Add support for timer

fensoft, 2023-01-19:
Like a timer button, when "on message" is sent, the output will be set to off x sec later.
If another "on message" is sent, timer is reset.

ptvoinfo, 2023-01-27:
@fensoft Do you mean the "On" message when the timer is still active?

fensoft, 2023-01-28:
Yes.

If time is 60s:

Press on-> wait 60s->Auto off
Press on->wait 30s->press on->wait 60s->auto off

ice2009, 2023-02-07:
Hello it possible make timer how ask user on up? for the switch and gpio ?

ptvoinfo, 2023-02-08:
@ice2009 Sorry, but I didn't understand your question.

ice2009, 2023-02-08:
i See you add timer to pulse switch . Switch turn off automaticly. It possible change seconds to miliseconds?

ptvoinfo, 2023-02-08:
@ice2009 No, this timer can measure seconds only. You can implement short timers using the "trigger" command (https://ptvo.info/zigbee-configurable-firmware-features/gpio/).

closed 2023-02-28 by ptvoinfo