ptvoinfo/zigbee-configurable-firmware #176: Firmware 1.8.0.7 not working with CC2652RB

CSRLabs, 2023-01-28:
Latest firmware 2023-01-12 ver. 1.8.0.7 seems to break support for CC2652RB device.
The same hardware and same configuration works fine with previous firmware 2022 1.7.3.0.

Symptoms: the status led does not turn on and the UART tx output is fixed at logic state 1

[image]

deaxx, 2023-01-28:
connect the quartz to the module and flash the firmware for cc2652r
[image]

CSRLabs, 2023-01-28:
@deaxx so you mean that newer firmware requires an external quartz to work?
This is not clear to me.
Please notice that previous firmware version works fine without any external quartz.

deaxx, 2023-01-28:
this is a way to make the module work on this particular version

CSRLabs, 2023-01-28:
I'll wait for the next fw version and hope this will be fixed.

ptvoinfo, 2023-02-08:
@CSRLabs Could you please test the latest release? I've resolved the issue with CC2652RB.

CSRLabs, 2023-02-08:
I confirm it is working fine now. Thank you for the quick fix.
I will keep the device monitored in the coming days just to make sure it is working as expected.

closed 2023-02-11 by ptvoinfo