ptvoinfo/zigbee-configurable-firmware #177: Motor + Mx1508

ice2009, 2023-02-06:
Hello :) I want create battery power blinds for window. I like danalock function because danalock sleep but when i lock door dana automatically door lock. I want  this function chip sleep and when a i send mqtt command from hassio to open chip read this command and open or close my blinds. I have problem when i set gpio for control motor driver mx1508 see this picture. Can you help me with my problem ?
[image] 
[image]
[image]

ptvoinfo, 2023-02-06:
This is just a warning. If you're confident that it is the correct setup, you can continue and save a firmware.

Usually, when you set an output "high", it causes high current leakage through this output. If your battery has a large capacity, or you set this output for a short time, you can use it.

ice2009, 2023-02-07:
Okay and could you help me with probblem with mac indirect timeout ? After i piar my diy device to sonoff zbbridge divece works perfectly but after 1-2 minute after pairing zigbee device go to sleep and no wake-up. This is my config from your ptvo program 

```
> Board type: CC2530
Device type: PSM
Update firmware's timestamp : 2023-02-07
Wait for commands after a report in PSM: Yes
Poll for queued commands (s): 10
Status LED: P04, Periodic (every 5 seconds)
Set default reporting interval (s): 60

Output pins:
P12: Output 1, GPIO, External pull-up (Role: Generic)
P13: Output 2, GPIO, External pull-up (Role: Generic)

Input pins:
P07: Input 1, External wake-up, Pull-up
P14: Input 2, GPIO, External pull-up, Switch (Bind command: On/Off)

```

and this is log from sonoff zbbridge after pairing 

```
12:35:14.969 ZIG: {"ZbEZSPReceived":"450000000021800000000100000544AD4A14FFFF02060002"}
12:35:14.973 MQT: tele/tempsen/RESULT = {"ZbBind":{"Device":"0x144A","Status":0,"StatusMessage":"SUCCESS"}}
12:35:16.637 ZIG: auto-bind `ZbBind {"Device":"0x144A","Endpoint":1,"Cluster":"0x0006"}`
12:35:16.638 ZIG: ZbEZSPSend 3400004A140000210000004001000007011607A9FC5E16004B1200010600031F5EE1FEFF15CD0401
12:35:16.650 ZIG: {"ZbEZSPReceived":"340000A9"}
12:35:16.824 ZIG: {"ZbEZSPReceived":"3F00004A1400002100000040010000A9010000"}
12:35:16.839 ZIG: {"ZbEZSPReceived":"450000000021800000000100000658B24A14FFFF02070002"}
12:35:16.843 MQT: tele/tempsen/RESULT = {"ZbBind":{"Device":"0x144A","Status":0,"StatusMessage":"SUCCESS"}}
12:35:18.442 ZIG: auto-bind `ZbSend {"Device":"0x144A","Config":{"Power":{"MinInterval":1,"MaxInterval":3600}}}`
12:35:18.444 ZigbeeZCLSend device: 0x144A, endpoint:1, cluster:0x0006, cmd:0x06, send:"000000100100100E"
12:35:18.445 ZIG: ZbEZSPSend 3400004A140401060001014001000004010B100406000000100100100E
12:35:18.497 ZIG: {"ZbEZSPReceived":"340000AA"}
12:35:19.091 ZIG: {"ZbEZSPReceived":"45000004010600010100010000073CAB4A14FFFF041804070002"}
12:35:19.093 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":"0x0006","srcaddr":"0x144A","srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"LinkQuality":6,"securityuse":0,"seqnumber":7,"fc":"0x18","frametype":0,"direction":1,"disableresp":1,"manuf":"0x0000","transact":4,"cmdid":"0x07","payload":"00"}}
12:35:19.095 ZIG: ZbZCLRawReceived: {"0x144A":{"ConfigResponse":{},"Endpoint":1,"LinkQuality":6}}
12:35:19.100 MQT: tele/tempsen/144A/SENSOR = {"0x144A":{"Device":"0x144A","ConfigResponse":{},"Endpoint":1,"LinkQuality":6}}
12:35:19.131 ZIG: {"ZbEZSPReceived":"3F00004A1404010600010140010000AA010000"}
12:35:20.410 ZIG: auto-bind `ZbBind {"Device":"0x144A","Endpoint":2,"Cluster":"0x0006"}`
12:35:20.411 ZIG: ZbEZSPSend 3400004A140000210000004001000008011608A9FC5E16004B1200020600031F5EE1FEFF15CD0401
12:35:20.466 ZIG: {"ZbEZSPReceived":"340000AB"}
12:35:22.418 ZIG: auto-bind `ZbSend {"Device":"0x144A","Config":{"Power":{"MinInterval":1,"MaxInterval":3600}}}`
12:35:22.420 ZigbeeZCLSend device: 0x144A, endpoint:2, cluster:0x0006, cmd:0x06, send:"000000100100100E"
12:35:22.420 ZIG: ZbEZSPSend 3400004A140401060001024001000005010B100506000000100100100E
12:35:22.471 ZIG: {"ZbEZSPReceived":"340000AC"}
12:35:22.690 ZIG: {"ZbEZSPReceived":"3F00004A1404010600010240010000AC010000"}
12:35:22.710 ZIG: {"ZbEZSPReceived":"450000040106000201000100000A38AA4A14FFFF041805070002"}
12:35:22.712 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":"0x0006","srcaddr":"0x144A","srcendpoint":2,"dstendpoint":1,"wasbroadcast":0,"LinkQuality":3,"securityuse":0,"seqnumber":10,"fc":"0x18","frametype":0,"direction":1,"disableresp":1,"manuf":"0x0000","transact":5,"cmdid":"0x07","payload":"00"}}
12:35:22.714 ZIG: ZbZCLRawReceived: {"0x144A":{"ConfigResponse":{},"Endpoint":2,"LinkQuality":3}}
12:35:22.718 MQT: tele/tempsen/144A/SENSOR = {"0x144A":{"Device":"0x144A","ConfigResponse":{},"Endpoint":2,"LinkQuality":3}}
12:35:23.844 ZIG: {"ZbEZSPReceived":"3F00004A1400002100000040010000A8010000"}
12:35:28.566 ZIG: {"ZbEZSPReceived":"450000040101000101000100000B38AA4A14FFFF0B18000A210020C82000202202"}
12:35:28.568 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":"0x0001","srcaddr":"0x144A","srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"LinkQuality":3,"securityuse":0,"seqnumber":11,"fc":"0x18","frametype":0,"direction":1,"disableresp":1,"manuf":"0x0000","transact":0,"cmdid":"0x0A","payload":"210020C820002022"}}
12:35:28.570 ZIG: ZbZCLRawReceived: {"0x144A":{"0001/0021":200,"0001/0020":34,"Endpoint":1,"LinkQuality":3}}
12:35:28.575 ZIG: {"ZbEZSPReceived":"450000040106000101000100000C3CAB4A14FFFF0718010A0000100002"}
12:35:28.577 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":"0x0006","srcaddr":"0x144A","srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"LinkQuality":6,"securityuse":0,"seqnumber":12,"fc":"0x18","frametype":0,"direction":1,"disableresp":1,"manuf":"0x0000","transact":1,"cmdid":"0x0A","payload":"00001000"}}
12:35:28.579 ZIG: ZbZCLRawReceived: {"0x144A":{"0006/0000":0,"Endpoint":1,"LinkQuality":6}}
12:35:28.586 ZIG: {"ZbEZSPReceived":"450000040106000201000100000D34A94A14FFFF0718020A0000100002"}
12:35:28.588 ZIG: {"ZbZCLReceived":{"groupid":0,"clusterid":"0x0006","srcaddr":"0x144A","srcendpoint":2,"dstendpoint":1,"wasbroadcast":0,"LinkQuality":0,"securityuse":0,"seqnumber":13,"fc":"0x18","frametype":0,"direction":1,"disableresp":1,"manuf":"0x0000","transact":2,"cmdid":"0x0A","payload":"00001000"}}
12:35:28.590 ZIG: ZbZCLRawReceived: {"0x144A":{"0006/0000":0,"Endpoint":2,"LinkQuality":0}}
[... cut ...]
12:38:55.151 MQT: tele/tempsen/RESULT = {"ZbRouteError":{"ShortAddr":"0x144A","Status":66,"StatusMessage":"MAC_INDIRECT_TIMEOUT"}}
12:38:55.345 ZIG: {"ZbEZSPReceived":"8000424A14"}
12:38:55.352 MQT: tele/tempsen/RESULT = {"ZbRouteError":{"ShortAddr":"0x144A","Status":66,"StatusMessage":"MAC_INDIRECT_TIMEOUT"}}
12:38:55.988 ZIG: {"ZbEZSPReceived":"45000000003880000040010000B0FF000000FFFF1B000000F8FF071A00070010FBA9DFB7B5ACD2B0AEA8AFA7ADB0A5BC"}
12:38:56.811 ZIG: {"ZbEZSPReceived":"3F00004A1404010600010140010000AE016600"}
12:38:56.940 ZIG: {"ZbEZSPReceived":"3F00004A1404010600010140010000AF016600"}

```

ptvoinfo, 2023-02-07:
In your case, the device should wake-up and check for commands every 10 seconds (the "Poll for queued command" option). 

Probably, if you send a command when the device sleeps, you can get a command timeout. Usually, a coordinator should queue a command for a PSM device for more seconds, and resend it to the device when it asks for queued data.

ice2009, 2023-02-07:
Okay and could you help me solve this problem? if i start use z2m in home-assistant its maybe solve this problem ?

ptvoinfo, 2023-02-07:
Z2M can queue commands for PSM device. You can try it. Maybe, it will work better in your case.