ptvoinfo/zigbee-configurable-firmware #178: Model ID does not update

gmeik, 2023-02-11:
The Model ID remains 'ptvo.switch', also after entering a custom Model ID.
I flash the CC2530 using an Arduino with CCLoader software (after the CCDebugger was fried - delivery time of a new one is quite long). After flashing, I can operate digital outputs without problems. Using custom MQTT commands, I can also change PWM levels. So the device seems to have been flashed properly. But because the Model ID doesn't update after reprogramming/repairing, the custom converter for Zigbee2MQTT is not activated.
Let me know if I need to supply more information. Or if the CCLoader approach is the likely cause of this trouble...

ptvoinfo, 2023-02-12:
@gmeik You should delete device in Z2M, restart Z2M and rejoin your device again. Z2M should display the new ModelId.

gmeik, 2023-02-13:
I tried deleting the device from Z2M, restart Z2M and rejoin. It still shows up as ptvo.switch.
I checked the .bin file and the 'ptvo.switch' text does not even show up, and my own model ID does show up. So for now I'll assume that programming with CCLoader didn't work correctly, and wait for the new CCDebugger.

closed 2023-02-13 by gmeik

ptvoinfo, 2023-02-17:
You can try to read the firmware back from the broken device and compare with the flashed HEX. The flasher uses a simple checksum checking method. It may fail in some rare cases.

phalox, 2023-09-02:
I seem to be experiencing the same issue:

[image]

[image]

I've deleted the device in Z2M, and restarted Z2M. None of it seems to fix the problem. When I make changes to the firmware (e.g. add an input), they do show up in Z2M.

I've also let it validate that the hex file is correct, and that seems to be the case:

[image]

I bumped into one more thing.... When exporting the converter and opening the file, I still see the old name (ptvo.switch).

[image]

Could it be that this is part of the issue?