ptvoinfo/zigbee-configurable-firmware #179: Device won't join Zigbee2MQTT but new device works perfectly

mortenmoulder, 2023-02-15:
So I am not entirely sure what's going on here. Zigbee2MQTT might be caching something, but I have removed my device completely, checked configuration.yaml, database.db, and any backup files there might be - not a single file contains my device's IEEE address except the log files.

Steps to reproduce:

- Remove the device from Z2M
- Restart Z2M
- Grab a brand new chip
- Flash the firmware to the new chip
- Tt joins perfectly

My custom device converter even works and tells me it's the correct Manufacturer name and Model ID (based on `zigbeeModel`, `model`, and `vendor`). The old one just says unknown and fails to interview.

Here's the log of the device that doesn't work (it just says interview failed):

```
Zigbee2MQTT:info 2023-02-15 02:08:41Device '0x00124b002a5f326d' joined
Zigbee2MQTT:info 2023-02-15 02:08:41MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x00124b002a5f326d","ieee_address":"0x00124b002a5f326d"},"type":"device_joined"}'
Zigbee2MQTT:info 2023-02-15 02:08:41MQTT publish: topic 'zigbee2mqtt/0x00124b002a5f326d/availability', payload 'online'
Zigbee2MQTT:info 2023-02-15 02:08:41MQTT publish: topic 'zigbee2mqtt/0x00124b002a5f326d', payload '{"device":{"friendlyName":"0x00124b002a5f326d","ieeeAddr":"0x00124b002a5f326d","model":"unknown","networkAddress":60825,"type":"Unknown"},"last_seen":"2023-02-15T02:08:40+01:00"}'
Zigbee2MQTT:info 2023-02-15 02:08:41Starting interview of '0x00124b002a5f326d'
Zigbee2MQTT:info 2023-02-15 02:08:41MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x00124b002a5f326d","ieee_address":"0x00124b002a5f326d","status":"started"},"type":"device_interview"}'
```

Here is the log of the device that works:

```
Zigbee2MQTT:info 2023-02-15 01:59:25: Device '0x00124b002a5ed5e4' joined
Zigbee2MQTT:info 2023-02-15 01:59:25: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x00124b002a5ed5e4","ieee_address":"0x00124b002a5ed5e4"},"type":"device_joined"}'
Zigbee2MQTT:info 2023-02-15 01:59:25: MQTT publish: topic 'zigbee2mqtt/0x00124b002a5ed5e4/availability', payload 'online'
Zigbee2MQTT:info 2023-02-15 01:59:25: MQTT publish: topic 'zigbee2mqtt/0x00124b002a5ed5e4', payload '{"device":{"friendlyName":"0x00124b002a5ed5e4","ieeeAddr":"0x00124b002a5ed5e4","model":"unknown","networkAddress":9436,"type":"Unknown"},"last_seen":"2023-02-15T01:59:25+01:00"}'
Zigbee2MQTT:info 2023-02-15 01:59:25: Starting interview of '0x00124b002a5ed5e4'
Zigbee2MQTT:info 2023-02-15 01:59:25: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x00124b002a5ed5e4","ieee_address":"0x00124b002a5ed5e4","status":"started"},"type":"device_interview"}'
Zigbee2MQTT:info 2023-02-15 01:59:26: MQTT publish: topic 'zigbee2mqtt/0x00124b002a5ed5e4', payload '{"device":{"friendlyName":"0x00124b002a5ed5e4","ieeeAddr":"0x00124b002a5ed5e4","model":"unknown","networkAddress":9436,"type":"Unknown"},"last_seen":"2023-02-15T01:59:26+01:00"}'
Zigbee2MQTT:debug 2023-02-15 01:59:26: Device '0x00124b002a5ed5e4' announced itself
Zigbee2MQTT:info 2023-02-15 01:59:26: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x00124b002a5ed5e4","ieee_address":"0x00124b002a5ed5e4"},"type":"device_announce"}'
Zigbee2MQTT:debug 2023-02-15 01:59:40: Received Zigbee message from '0x00124b002a5ed5e4', type 'readResponse', cluster 'genBasic', data '{"modelId":"ZIG-SW-3"}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2023-02-15 01:59:40: Skipping message, definition is undefined and still interviewing
Zigbee2MQTT:info 2023-02-15 01:59:40: MQTT publish: topic 'zigbee2mqtt/0x00124b002a5ed5e4', payload '{"device":{"friendlyName":"0x00124b002a5ed5e4","ieeeAddr":"0x00124b002a5ed5e4","manufacturerID":4447,"model":"unknown","networkAddress":9436,"type":"EndDevice"},"last_seen":"2023-02-15T01:59:40+01:00","linkquality":138}'
Zigbee2MQTT:debug 2023-02-15 01:59:40: Received Zigbee message from '0x00124b002a5ed5e4', type 'readResponse', cluster 'genBasic', data '{"manufacturerName":"smarthjemmet.dk"}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2023-02-15 01:59:40: Skipping message, definition is undefined and still interviewing
Zigbee2MQTT:info 2023-02-15 01:59:40: MQTT publish: topic 'zigbee2mqtt/0x00124b002a5ed5e4', payload '{"device":{"friendlyName":"0x00124b002a5ed5e4","ieeeAddr":"0x00124b002a5ed5e4","manufacturerID":4447,"manufacturerName":"smarthjemmet.dk","model":"unknown","networkAddress":9436,"type":"EndDevice"},"last_seen":"2023-02-15T01:59:40+01:00","linkquality":138}'
Zigbee2MQTT:debug 2023-02-15 01:59:40: Received Zigbee message from '0x00124b002a5ed5e4', type 'readResponse', cluster 'genBasic', data '{"powerSource":3}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2023-02-15 01:59:40: Skipping message, definition is undefined and still interviewing
Zigbee2MQTT:info 2023-02-15 01:59:40: MQTT publish: topic 'zigbee2mqtt/0x00124b002a5ed5e4', payload '{"device":{"friendlyName":"0x00124b002a5ed5e4","ieeeAddr":"0x00124b002a5ed5e4","manufacturerID":4447,"manufacturerName":"smarthjemmet.dk","model":"unknown","networkAddress":9436,"powerSource":"Battery","type":"EndDevice"},"last_seen":"2023-02-15T01:59:40+01:00","linkquality":138}'
Zigbee2MQTT:debug 2023-02-15 01:59:40: Received Zigbee message from '0x00124b002a5ed5e4', type 'readResponse', cluster 'genBasic', data '{"zclVersion":1}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2023-02-15 01:59:40: Skipping message, definition is undefined and still interviewing
Zigbee2MQTT:info 2023-02-15 01:59:40: MQTT publish: topic 'zigbee2mqtt/0x00124b002a5ed5e4', payload '{"device":{"friendlyName":"0x00124b002a5ed5e4","ieeeAddr":"0x00124b002a5ed5e4","manufacturerID":4447,"manufacturerName":"smarthjemmet.dk","model":"unknown","networkAddress":9436,"powerSource":"Battery","type":"EndDevice","zclVersion":1},"last_seen":"2023-02-15T01:59:40+01:00","linkquality":138}'
Zigbee2MQTT:debug 2023-02-15 01:59:40: Received Zigbee message from '0x00124b002a5ed5e4', type 'readResponse', cluster 'genBasic', data '{"appVersion":1}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2023-02-15 01:59:40: Skipping message, definition is undefined and still interviewing
Zigbee2MQTT:info 2023-02-15 01:59:40: MQTT publish: topic 'zigbee2mqtt/0x00124b002a5ed5e4', payload '{"device":{"applicationVersion":1,"friendlyName":"0x00124b002a5ed5e4","ieeeAddr":"0x00124b002a5ed5e4","manufacturerID":4447,"manufacturerName":"smarthjemmet.dk","model":"unknown","networkAddress":9436,"powerSource":"Battery","type":"EndDevice","zclVersion":1},"last_seen":"2023-02-15T01:59:40+01:00","linkquality":138}'
Zigbee2MQTT:debug 2023-02-15 01:59:40: Received Zigbee message from '0x00124b002a5ed5e4', type 'readResponse', cluster 'genBasic', data '{"stackVersion":1}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2023-02-15 01:59:40: Skipping message, definition is undefined and still interviewing
Zigbee2MQTT:info 2023-02-15 01:59:40: MQTT publish: topic 'zigbee2mqtt/0x00124b002a5ed5e4', payload '{"device":{"applicationVersion":1,"friendlyName":"0x00124b002a5ed5e4","ieeeAddr":"0x00124b002a5ed5e4","manufacturerID":4447,"manufacturerName":"smarthjemmet.dk","model":"unknown","networkAddress":9436,"powerSource":"Battery","stackVersion":1,"type":"EndDevice","zclVersion":1},"last_seen":"2023-02-15T01:59:40+01:00","linkquality":138}'
Zigbee2MQTT:debug 2023-02-15 01:59:40: Received Zigbee message from '0x00124b002a5ed5e4', type 'readResponse', cluster 'genBasic', data '{"hwVersion":1}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2023-02-15 01:59:40: Skipping message, definition is undefined and still interviewing
Zigbee2MQTT:info 2023-02-15 01:59:40: MQTT publish: topic 'zigbee2mqtt/0x00124b002a5ed5e4', payload '{"device":{"applicationVersion":1,"friendlyName":"0x00124b002a5ed5e4","hardwareVersion":1,"ieeeAddr":"0x00124b002a5ed5e4","manufacturerID":4447,"manufacturerName":"smarthjemmet.dk","model":"unknown","networkAddress":9436,"powerSource":"Battery","stackVersion":1,"type":"EndDevice","zclVersion":1},"last_seen":"2023-02-15T01:59:40+01:00","linkquality":138}'
Zigbee2MQTT:debug 2023-02-15 01:59:41: Received Zigbee message from '0x00124b002a5ed5e4', type 'readResponse', cluster 'genBasic', data '{"dateCode":"2023-02-15"}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2023-02-15 01:59:41: Skipping message, definition is undefined and still interviewing
[... cut ...]
The custom converter for Z2M is just a "Save custom converter for Z2M" and there are no changes. If I remove Manufacturer name and Model ID, so it's just a `ptvo.switch`, it also cannot join anymore.

I have tried pairing through a router instead of directly to my CC2652P coordinator.

Something tells me Z2M has compeltely banned this device from joining again, but the IEEE is no where to be found except in logs, and I did try to remove those logs.

Where do I go from here? I am currently debugging creating my own custom converter for Z2M, and it's not easy when Z2M refuses to interview my devices correctly after a few attempts (remove, restart Z2M, pair again).

ptvoinfo, 2023-02-15:
Hi!
1. What type of firmware did you flash (PSM, end-device, router)?
2. Did you try to reset your device to factory settings (by power cycle or using a system button)? https://ptvo.info/zigbee-configurable-firmware-features/general-info/

mortenmoulder, 2023-02-15:
@ptvoinfo I have tried both, but primarily my devices run PSM

Yes to both. My input 1 is set to pull up, so I can just short it to ground.

[image]

ptvoinfo, 2023-02-17:
Could you add a status LED to your device? It may help to diagnose this problem.
Maybe, you should wake up your PSM device while it joins to a coordinator by short clicks on Input1.

mortenmoulder, 2023-02-17:
@ptvoinfo There's already an LED, but it only lights up when "Joining or errors". It blinks with maybe 5 seconds intervals until it starts getting interviewed by Zigbee2MQTT, then it stops blinking if I recall correctly.

I have tried waking up the device with short clicks on Input 1. Maybe 1 second delay between each. It simply doesn't matter :(

ptvoinfo, 2023-02-18:
@mortenmoulder
If the LED does not blink, the device thinks that it is connected to a coordinator. Could you try to make a new separate network with a coordinator and single device?

closed 2023-05-06 by ptvoinfo