ptvoinfo/zigbee-configurable-firmware #180: Expose `link_to_output` as `switch` and not as `enum`

Danit2, 2023-02-17:
It woud be better to expose the option `link_to_output` as a switch in Z2M.
Beqause it is easyer to handle this in Home Assistant.

I have try it to change the external converter from
```
features.push(exposes.enum('link_to_output', exposes.access.ALL,
        getSortedList(inputLinkList)).withEndpoint(epName));
```
to
```
features.push(exposes.switch().withState('link_to_output', false, '', exposes.access.ALL, 'ON', 'OFF').withEndpoint(epName));
```
With this it works but it comes not right in the frontend of Z2M and Home Assistant. 
In Z2M it comes in the fronted under the state but without the endpointname.
In Home Assistant it comes as `switch.friendly_name` and not as `switch.friendly_name_link_to_output_l2`

Can you help to write the converter right? 

Thanks.

ptvoinfo, 2023-02-27:
It is difficult to change this option because it is hard-coded in the Z2M web interface.

Danit2, 2023-02-27:
Ahh ok thanks for your answer. 

I have found a solution. 

I have changed the auto discovery from zigbee2mqtt to homeassistant in the device option with this: 
```
homeassistant:
    link_to_output_l2:
      type: switch
      options: null
      payload_off: 'no'
      payload_on: 'yes'
```

Now i have a switch in homeassistant.

closed 2023-02-28 by ptvoinfo