ptvoinfo/zigbee-configurable-firmware #182: Use Sonoff SNZB-01 for bistable swich

robotsflo, 2023-02-21:
Hello,
Did you think that is possible to flash with custom firmware a cheap Sonoff SNZB-01 push button to do the same functionnality that an [hue wall switch](https://www.philips-hue.com/en-gb/p/hue-philips-hue-wall-switch-module/8719514318045) 
I would like to send "Toggle" command each time that i press on a bistable rocker switch 
Most of manufactured device use basic push button. 
Today, in my experience, only the "hue wall switch" have this feature but is expensive.

Thanks a lot for your reply.

[image]
[image]
[image]

ptvoinfo, 2023-02-24:
@robotsflo 
You can try to select "Switch" + "Toggle" + Bind command "Toggle" for "Input".

robotsflo, 2023-02-27:
Info : 
P1.0 : led pin 
P1.5 : Main push button
P2.0 : Reset push button
[image]

closed 2023-05-06 by ptvoinfo