ptvoinfo/zigbee-configurable-firmware #183: e18-ms1-pcb + bme280 PSM

bachy, 2023-02-26:
Hi,

i'm trying to setup a temp sensor PSM end device with e18-ms1-pcb chip coupled with a bme280 but current consumption stay way to high: 44mA in active and 16ma between reports

I can't figure out why

here are my ptvo conf, schematics and pcb design
[image]
[image]
[image]
[image]

any idea ?

ptvoinfo, 2023-02-27:
@bachy Hi, I do not see external pull-up resistors in your schematic.

bachy, 2023-02-27:
Hi, thank you for your answer !

this is my bme280 board, it has pullups, right ?
[image]

i powered the bme280 board directly to vcc and disabled "external sensor power" 
[image]
[image]

it is better but i steel have 27mA active and 200µA in power saving mode, wich seems still to much to me, no ?
i should have only a few µA when sleeping ?

ptvoinfo, 2023-02-27:
OK. Could you check power consumption without BME attached? Maybe, there is a current leak in another place?

bachy, 2023-02-27:
bme280 totaly detached, the circuit consume 1µA

bachy, 2023-02-27:
ok, it seems to be good, i re-attached the bme, and once repaired to z2m it consume 1.1µA in sleepmode and 8mA when sending data

all goods ! thanks

could you explained why the "external power sensor was not working ?

ptvoinfo, 2023-02-28:
It works with some sensors. In some cases, it requires a special power control circuit. 
You can also try this output type with internal pull-up or pull-down resistors.

closed 2023-03-22 by ptvoinfo