ptvoinfo/zigbee-configurable-firmware #186: Please add XH711 sensor.  Vote who needs it too.

ikey-ru, 2023-03-08:
Рассмотрите пожалуйста возможность добавить популярную микросхему для тензодатчиков XH711.
Год назад Philip тоже просил:
[image]

Кому тоже нужно, пожалуйста, проголосуйте.

dido8884, 2023-03-10:
Да, идея нормальная, к этим платам можно подключить датчик давления резистивный.

Boolkinator, 2023-03-10:
👍

Garkus98, 2023-03-11:
+

cat-inGlasses, 2023-03-20:
👍

pxbiuibiu, 2023-03-21:
👍

LimeQuat, 2023-03-22:
+

epvdvalk, 2023-03-28:
Would like that!

darkside9009, 2023-05-15:
please add HX711 :-)

francisbesset, 2023-05-28:
@ptvoinfo I see that HX711 is available for CC2652 and CC1352 only.
Do you think will be available in further for CC2530?

Thank you very much for your work.

ptvoinfo, 2023-05-29:
@francisbesset There is no free flash memory for new features. Therefore, I add them for new chips (CC2652 and CC1352) only.

closed 2023-06-02 by ptvoinfo