ptvoinfo/zigbee-configurable-firmware #187: Конвертер для счетчика модкам ( поле корректировки wakeup interval )

DaN66O, 2023-03-10:
Можно ли сделать в конвертере отдельное поле для корректировки  wakeup interval ?
Зачем в двух полях в каждом счетчике да еще с минусом вводить ?
Для интервала достаточно одного поля ввода.
В z2m  затруднительно вводить отрицательные числа в этом поле.
Приходится вводить положительное число, а потом добавлять минус. 
Несколько раз видел сообщение что отрицательные числа не допускаются. 
Видимо что то обновилось в новых версиях з2м.

DaN66O, 2023-05-05:
Я повторю свою просьбу....

ptvoinfo, 2023-05-06:
Пожалуйста, попробуйте.
[ptvo_counter_2ch.zip](https://github.com/ptvoinfo/zigbee-configurable-firmware/files/11411794/ptvo_counter_2ch.zip)

DaN66O, 2023-05-06:
Данная опция пропала.
Как теперь вводить масштабный коэффициент  ?
[image]

DaN66O, 2023-05-07:
Дополнение...
Не работает wakeup interval
в логе следующие ошибки
[image]

ptvoinfo, 2023-05-09:
The converter final version.
[ptvo_counter_2ch.zip](https://github.com/ptvoinfo/zigbee-configurable-firmware/files/11427731/ptvo_counter_2ch.zip)