ptvoinfo/zigbee-configurable-firmware #190: Zigbee2mqtt could not search for PTVO devices

24021438, 2023-03-21:
I have an E18-TBH-27 and its chip is E18-2G4Z27SI, I can't search it in zigbee2mqtt after flashing, do I need to put js in to search the device? It still doesn't ne
[image]
[image]
https://www.ebyte.com/product-view-news.html?id=629

24021438, 2023-03-21:
[image]
[image]

ptvoinfo, 2023-03-22:
1. Enable any GPIO output.
2. Flash the update firmware. The status LED should blink every second when the device tries to join.
3. Z2M should detect a new joined device.
If it does not work, you may try CC2530+CC2591/CC2592 device types instead.

turboproc, 2023-03-26:
Using this firmware a lot these days and so far no issues in binding with Z2M. Could you share the .js file? It's crucial to have that right and also make sure it's incude in the configuration.yaml file otherwise it will not be loaded. In that file you'll to have a line like
```
external_converters:
  - ptvo_temp.js
```
of course have it adjusted for your case.

24021438, 2023-03-31:
> 

Now it's okay because aduino didn't refresh CC2530 normally, I used CCdebugger again, and now I have questions about CCT, have you ever done CCT

closed 2023-05-06 by ptvoinfo