ptvoinfo/zigbee-configurable-firmware #198: PSM don't wake-up

axelhorn, 2023-04-20:
What am I doing wrong?
Sonoff SNZB-04 module (on ZB-DS01 v1.0 board) - simple door/window sensot with CC2530F256 chipset.
When I load the "end device" firmware to it - it works properly. When I load PSM, I force it to pair with a reset - the LED flashes from time to time for a few seconds, then it goes into a deep sleep state and cannot be woken up in any way. external wake-up, neither a sensor nor periodic reports.It doesn't matter what I set in the advanced options, after falling asleep the module stops responding.
However, if after pairing I press the external wake-up (input1) button every few seconds, preventing him from falling asleep, the module flashes but does not pair.

[image]
[image]

ptvoinfo, 2023-04-21:
@axelhorn 
Sorry, but I don't know anything about a schematic of this device. Are you sure about "External pull-up" and "Inversed" for "Input 1"?

axelhorn, 2023-04-21:
Yes, GPIO 2_0 is pulled up by resistor to +battery and button switches do ground.

ptvoinfo, 2023-04-21:
Then you should disable "Inversed" for "Input 1".

axelhorn, 2023-04-22:
No combination helps.
However, reading the documentation from the CC2530, I see that a 32kHz crystal oscillator is needed on the XOSC32K inputs to wake up from sleep mode by the timer. Can you confirm if the oscillator is necessary for your PSM firmware or if I understand it correctly because 
I'm not sure. Because this Sonoff module does not have such an oscillator. It only has the main 32MHz.
[image]

deaxx, 2023-08-03:
> No combination helps. However, reading the documentation from the CC2530, I see that a 32kHz crystal oscillator is needed on the XOSC32K inputs to wake up from sleep mode by the timer. Can you [...]

Use this config
[image]

axelhorn, 2023-08-08:
Alleluyah! :) It's working. Sonoff ZB-DS01 open door sensor works with this configuration. Additionally, so far it seems to run more stable on the ptvo firmware than on the original Sonoff firmware.

closed 2023-08-10 by axelhorn