ptvoinfo/zigbee-configurable-firmware #201: big OTA won't start

axelhorn, 2023-04-25:
Hi, PTVO.
I have another problem.
I added the MX25R8035FZUIL0 flash memory to one of my Zigbbe devices. I uploaded new firmware with OTA. After rebooting the device, the LED blinks rapidly for a few seconds indicating the firmware is being copied to flash. Service command 131 returns the correct signature "C2 28 14" in attribute 16385. The device is working properly.
I have prepared a full firmware update via OTA according to the instructions on the website. After trying to upload for several dozen seconds on the Zigbbee@MQTT I can see an empty progress bar and "estimated time: N/A". Then there is an error in the logs:

```
2023-04-25 22:25:24Received MQTT message on 'z2m/bridge/request/device/ota_update/update' with data '{"id":"0x00124b00257934ab","transaction":"a0678-23"}'
info 2023-04-25 22:25:24Updating '0x00124b00257934ab' to latest firmware

```
`2023-04-25 22:25:25Update of '0x00124b00257934ab' at 0.00%`

```
info  2023-04-25 22:27:53: MQTT publish: topic 'z2m/0x00124b00257934ab', payload'{"bind_command_l7":null,"bind_command_l8":null,"contact_l7":false,"contact_l8":true,"device":"applicationVersion":1,"dateCode":"2023-04-25","friendlyName":"0x00124b00257934ab","hardwareVersion":1,"ieeeAddr":"0x00124b00257934ab","manufacturerID":0,"manufacturerName":"ARTi","model":"ptvo.GWC-E72","networkAddress":27045,"powerSource":"Mains (single phase)","softwareBuildID":"2023-04-25","stackVersion":3,"type":"Router","zclVersion":1},"l1":20,"last_seen":"2023-04-25T22:27:49+02:00","link_to_output_l7":null,"link_to_output_l8":null,"linkquality":211,"state_l2":"OFF","state_l3":"OFF","state_l4":"OFF","state_l5":"OFF","switch_actions_l7":null,"switch_actions_l8":null,"switch_type_l7":null,"switch_type_l8":null,"update":{"state":"available"},"update_available":null}'
info  2023-04-25 22:27:53: MQTT publish: topic 'z2m/bridge/response/device/ota_update/update', payload '{"data":{"id":"0x00124b00257934ab"},"error":"Update of '0x00124b00257934ab' failed (Timeout: device did not request any image blocks)","status":"error","transaction":"a0678-23"}'
error 2023-04-25 22:27:53: Update of '0x00124b00257934ab' failed (Timeout: device did not request any image blocks)
debug 2023-04-25 22:27:53: Error: Timeout: device did not request any image blocks
    at /app/node_modules/zigbee-herdsman-converters/lib/ota/common.js:360:28
    at runMicrotasks (<anonymous>)
    at runNextTicks (node:internal/process/task_queues:61:5)
    at listOnTimeout (node:internal/timers:528:9)
    at processTimers (node:internal/timers:502:7)
```

But if I prepare a small OTA update (settings only), the device updates itself after a dozen or so seconds, reporting the correct update in the logs.

axelhorn, 2023-04-28:
No, it's regular router. I'll try some more tests.

axelhorn, 2023-05-01:
I think I found the reason why the device is not downloading the update even though the update is reported in the logs.
The cause is the watchdog enabled in the firmware installed on the device. After disabling the watchdog and installing such firmware, the traditional "via cable", the next large and small OTAs pass without problems. Previously, only small (configuration) OTAs were passed with the watchdog turned on, and not always.
PTVO - can you test it yourself and possibly confirm if it's just my problem with this one DIY device, or if a firmware fix is needed.

ptvoinfo, 2023-05-04:
@axelhorn I cannot repeat this problem with my test board.

axelhorn, 2023-05-06:
Hmmm.... It's amazing. I try it with another of my devices (PSM too) and write here results of tests. This one above still updates fine every time, but only when watchdog is disabled.

closed 2023-10-31 by ptvoinfo