ptvoinfo/zigbee-configurable-firmware #208: Not working after upgrade to Z2MQTT 1.31.0-1

OskarL83, 2023-06-15:
As described in https://github.com/Koenkk/zigbee2mqtt/issues/17931

ptvoinfo, 2023-06-20:
@OskarL83 Could you please try the latest "dev" version of Z2M?

OskarL83, 2023-11-02:
Is this fixed now?

ptvoinfo, 2023-11-03:
@OskarL83 I didn't receive any feedback on this and supposed that this problem was fixed. Moreover, the latest Z2M version contains improvements for PTVO.

OskarL83, 2023-11-03:
Ok, what release/version?
I have tested 1.33.1-1, and that one didn't work with my device.

ptvoinfo, 2023-11-03:
https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.33.2

OskarL83, 2023-11-03:
1.33.2 is a little better. I now can see the states on all switches. But it doesn't work to turn them on and off. :-(

ptvoinfo, 2023-11-05:
It is strange. Could you please attach a screenshot with your device configuration, switches in Z2M and the error message?

OskarL83, 2024-03-08:
Tried with Z2MQTT 1.36.0-1 now. Still not working when trying to change switch state. 
[image]
. :-(
See attached error. 
[image]

ptvoinfo, 2024-03-09:
@OskarL83 Try to click the "Reconfigure" button on the "About" tab. How old is your flashed device's firmware?

OskarL83, 2024-03-09:
It's the same after a reconfigure of the device.
Running firmware version
20200405. How do I upgrade the firmware without a programmer/flasher?
I'm using this 4-channel relay: https://www.tindie.com/products/a_lab_technology/zigbee-4-channel-relay/

ptvoinfo, 2024-03-09:
@OskarL83 Sorry, but you cannot upgrade the firmware without a programmer/flasher. But you can use Rapsberry PI or Arduino as flasher.

OskarL83, 2024-03-09:
So after upgrade it will work with new releases of Zigbee2mqtt?

OskarL83, 2024-03-31:
Hi, why closed?
After upgrade it will work with new releases of Zigbee2mqtt?

ptvoinfo, 2024-04-01:
If you can update the firmware and z2m to the latest version, and it will not work, I can check everything on my computer and resolve the issue (if any).

closed 2025-03-21 by ptvoinfo