ptvoinfo/zigbee-configurable-firmware #209: Where has the BME280 support gone?

axelhorn, 2023-06-22:
Today I installed version 1.9.0.3 of the configurator to generate firmware for the BME280 sensor, and I see that it is not in the available list of sensors. Just yesterday, when I was working on version 1.8.0, the sensor was available.
What happened to the handling of this sensor?

ptvoinfo, 2023-06-22:
Hi,

It was moved to the 'Temperature sensor I2C' sensor group.

axelhorn, 2023-06-22:
Ughhhh... I was afraid you deleted it completely :)
It would be great if you updated the changelog on the website more often (last update currently 2022-07-21) because sometimes you will do something interesting, and not everyone will notice it.
I salute you and thanks for your great work.

closed 2023-06-22 by axelhorn