ptvoinfo/zigbee-configurable-firmware #215: Release gets sent without hold first

mortenmoulder, 2023-07-26:
I configured a device as a regular multi purpose switch with single, double, triple and hold/release. As discussed on Telegram, the delay between click and "send hold to coordinator" is about 2000ms. I will use 2000ms as an example further on.

If I hold down a button, I would expect to see "hold" in my Zigbee coordinator after 2000ms. However, if I hold down the button for maybe 1900ms and let go, it will instead send a "release" signal. The "hold" never gets sent to the coordinator. Here's a timeline example:

<hr>

```
0ms: click button
2000ms: "send hold"
2500ms: release button
2600ms: "send release"
```

This is the intended behavior. I added a 100ms delay between releasing the button to device sending "release" signal, just as an example.

<hr>

```
0ms: click button
1500ms: release button
1600ms: "send single_1"
```

Again, intended behavior. I didn't hold the button long enough to trigger a hold, therefore hold/release never gets sent. This is expected.

<hr>

```
0ms: click button
1900ms: release button
2000ms: "send release"
```

<hr>

In this example, "hold" never gets sent to the coordinator, but "release" signal is sent instead. Because the time clicked is less than 2000ms (which is the hold delay as we discussed on Telegram), I would expect it to send "single_1" instead of "release". It should never be possible to send "release" without a "hold" first, unless the switch is configured as a switch and not multi purpose (single, double, triple).

Can you try to replicate and confirm if this is an issue? Here is a clear screenshot that shows Z2M is able to receive a "release" without a "hold" first:

[image]

ptvoinfo, 2023-08-01:
@mortenmoulder 
I've tested it on CC2530 and I cannot repeat this problem (if I press a button with a finger). Could you enable debug logs in z2m and check zigbee messages?

```
debug 2023-08-01 14:34:52Received Zigbee message from '0x00124b0022055177', type 'attributeReport', cluster 'genMultistateInput', data '{"presentValue":4}' from endpoint 1 with groupID 0
debug 2023-08-01 14:34:53Received Zigbee message from '0x00124b0022055177', type 'attributeReport', cluster 'genMultistateInput', data '{"presentValue":0}' from endpoint 1 with groupID 0
```

"presentValue":0 - release
"presentValue":4 - hold

It is a good idea to verify it using a sniffer. Maybe, these messages were sent by a device but not processed by a coordinator.

mortenmoulder, 2023-08-01:
@ptvoinfo What happens if you release it JUST before hold is supposed to be sent? It takes me about 10-15 tries to replicate this behavior. Depends how good I am at counting to 2 in seconds.

Release too early and it says single, release too late and it says hold+release, but release at the exact time, and it will just say release.

ptvoinfo, 2023-08-01:
@ptvoinfo I get single_1. But I cannot catch a moment as in your 3rd example.

obrain17, 2023-08-30:
I can reproduce this behaviour on a CC2652P with this Input configuration:

[image]

Input 2 sends _single_l2_, _double_l2_ for short clicks and after having hold the button longer than 2 seconds: _release_l2_. 
But **no _hold_l2_** in between. Only presentValue = 0 can be seen in both Z2M log and sniffer:

[image]

@mortenmoulder  I did several attempts with different time holding the button but cannot produce the behaviour that _hold_l2_ is sent ever.

Very interesting is the fact:
Input 1 does send _hold_l1_ when pressing longer than 2 seconds, and _release_l1_ afterwards.  (no _single_1_, _double_1_ ... but On/off state for Output 1 as expected for a linked input)
Shouldn't it send any multistate messages as it is configured as as `linked to Output 1` ? (according [GPIO inputs](https://ptvo.info/zigbee-configurable-firmware-features/gpio/) : 

> Link – The input immediately controls the linked output state (e.g., Input 1 – Output 1). The device sends the on/off state of the output to a coordinator. Otherwise, the input is independent.

ptvoinfo, 2023-09-12:
@obrain17 Please reset the device that works incorrectly (by switching power or by Input 1). Maybe, the device restores keys mode from NVRAM.

closed 2023-10-31 by ptvoinfo