ptvoinfo/zigbee-configurable-firmware #216: Something wrong with ADC in PSM mode

axelhorn, 2023-07-28:
Hi,
Something is wrong with the configuration of the ADCs in the PSM mode (I did not check in other modes) in the firmware version 1.9.0.3.
When I configure the input as ADC (max. 3.3V), after restarting the CC2652P for the first cycle until the device goes to sleep, the measurement is correct. Then, when the device goes to sleep, as if the ADC pin was set low or some pull-down, and it stays like that. With subsequent awakenings of the device, the CC2652P constantly forces the pin to be around 0V. After restart, one measurement is correct again and the next pin wake-ups are forced to 0V.
After downgrading to firmware 1.8.0.9 with the same firmware configuration everything works perfectly.

ptvoinfo, 2023-08-01:
@axelhorn, I cannot repeat this problem with my chip. Could you please show screenshots with your full configuration?

axelhorn, 2023-08-01:
Of course.
[image]

[image]

[image]

ptvoinfo, 2023-08-02:
I've tried your list of outputs and pins and I cannot repeat the problem again. Could you please try on your side to disable features one by one and find a reason for this behavior?

deaxx, 2023-08-03:
> Hi, Something is wrong with the configuration of the ADCs in the PSM mode (I did not check in other modes) in the firmware version 1.9.0.3. When I configure the input as ADC (max. 3.3V), after [...]

Check this config
[image]

axelhorn, 2023-08-03:
Thanx. I'll try is when I found some piece of free time :) and write results.

axelhorn, 2023-08-14:
@ptvo Finally I checked the configuration you provided with version 1.9.0.3 and latest 1.9.0.6. And it's a little different.

With these settings (External sensor power control first, then ADC) even without RF control and without Internal temperature - high state on the sensor power supply appears every time and the sensors are powered, but again - data from the sensors are read only in first cycle after module reset. With subsequent wake-up cycles, the firmware reports zero adjusted for Scale factor and Shift on the first ADC, and 0V on the second ADC.
Let me remind you that in version 1.8.0.9 everything worked in the original configuration without any problems - External power worked with every wake-up cycle and each time the ADC value was read.

`2023-08-14 23:26:53MQTT publish: topic 'z2m/GSMM4', payload '{"action":null,"battery":100,"bind_command_l1":null,"device":{"applicationVersion":1,"dateCode":"2023-08-14","friendlyName":"GSMM4","hardwareVersion":1,"ieeeAddr":"0x00124b0022ab7939","manufacturerID":0,"manufacturerName":"ARTi","model":"ptvo.GSMM-E72","networkAddress":9790,"powerSource":"Battery","softwareBuildID":"2023-08-14","stackVersion":3,"type":"EndDevice","zclVersion":1},"l2":0.413,"l3":0.047,"l6":4.025,"l7":2.206,"l8":11,"last_seen":"2023-08-14T23:26:51+02:00","link_to_output_l1":null,"linkquality":142,"switch_actions_l1":null,"switch_type_l1":null,"temperature_l8":11,"update":{"installed_version":539166504,"latest_version":539166740,"state":"available"},"update_available":null,"voltage":3200,"voltage_l2":0.4,"voltage_l3":0,"voltage_l6":4,"voltage_l7":2.2}''`

athtest800, 2023-08-16:
I have the same problem. First reading OK. After sleep, the readings are 0V.

ptvoinfo, 2023-08-17:
@athtest800 Could you please attach a screenshot with your firmware configuration?

athtest800, 2023-08-17:
I did also some debugging on versions. The problem started at version "1.9.0.4" and later. Please, check what changed between the problematic "1.9.0.4" and the working "1.9.0.2".
Here my settings:
[image]

axelhorn, 2023-08-17:
@athtest800 Can you provide me with version 1.9.0.2 for testing? Because I don't have it. The previous one I have 1.8.

closed 2023-11-02 by ptvoinfo

axelhorn, 2024-06-02:
The problem with the controller forcing the ADC input (any used one) to 0V immediately after the first work-sleep cycle remains unsolved until the current firmware version.
It occurs on both the CC2652P and CC2652RB, which I am currently creating a new device on.