ptvoinfo/zigbee-configurable-firmware #217: PCA9685 does not work on new version

athtest800, 2023-08-08:
Hi, I try to use PCA9685 in the new version 1.9.0.4 but does not work.
I tried an older version 1.6.2.4 and works without problems. The settings are the same. Is there a fix for this? Thank you

[image]

ptvoinfo, 2023-08-08:
Could you please try with to use "Pull-up" instead of "External pull-up"?

athtest800, 2023-08-08:
Nothing changed. I get this message when querying  status of pwm channel.
`Publish 'get' 'brightness' to '0x00124b001ccc5454' failed: 'Error: Read 0x00124b001ccc5454/4 genLevelCtrl(["currentLevel"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'`

I will install all the versions after 1.6.2.4, in order to inform you the version the problem started.

athtest800, 2023-08-08:
I checked all the versions.  The problem starts at version 1.9.0.2 and after. Can you please check and fix?

ptvoinfo, 2023-08-08:
@athtest800 Fixed in the latest version. Thank you for the reporting.

closed 2023-08-17 by ptvoinfo