ptvoinfo/zigbee-configurable-firmware #222: Some I2C sensors no longer available with latest firmware

glcos, 2023-08-13:
I noticed that some popular I2C sensors such as BMP280 and SHT30 are no longer available in the latest firmware.
Is this a bug or a design decision?

ptvoinfo, 2023-08-14:
@glcos 
These sensors were moved to a separate group:
https://ptvo.info/zigbee-configurable-firmware-features/external-sensors/i2c-sensor/

glcos, 2023-08-14:
Sensors BMP280 and SHT30 have disappeared from the GUI dropdown list

[image]

This is how the list looked like in an older version of the firmware GUI

[image]

glcos, 2023-08-14:
I managed to generate the firmware for my application. Thanks for your instructions.
However, it is worth noticing that latest GUI is unable to read previous configuration .INI files and Hex FW files when I2C sensors are defined.

glcos, 2023-08-15:
I updated the firmware on the actual device, it is working fine apart from the BMP280 sensor.
The temperature part of BMP280 is read OK but the pressure is always reporting a fixed value of 1000.

[image]

```
2023-08-15 15:28:06.323 DEBUG (MainThread) [zigpy.zcl] [0x8C4E:7:0x0402] Received ZCL frame: b'\x18\xfe\n\x00\x00)\x9a\n'
2023-08-15 15:28:06.324 DEBUG (MainThread) [zigpy.zcl] [0x8C4E:7:0x0402] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=0, direction=<Direction.Client_to_Server: 1>, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True), tsn=254, command_id=10, *direction=<Direction.Client_to_Server: 1>)
2023-08-15 15:28:06.325 DEBUG (MainThread) [zigpy.zcl] [0x8C4E:7:0x0402] Decoded ZCL frame: TemperatureMeasurement:Report_Attributes(attribute_reports=[Attribute(attrid=0x0000, value=TypeValue(type=int16s, value=2714))])
2023-08-15 15:28:06.325 DEBUG (MainThread) [zigpy.zcl] [0x8C4E:7:0x0402] Received command 0x0A (TSN 254): Report_Attributes(attribute_reports=[Attribute(attrid=0x0000, value=TypeValue(type=int16s, value=2714))])
2023-08-15 15:28:06.325 DEBUG (MainThread) [zigpy.zcl] [0x8C4E:7:0x0402] Attribute report received: measured_value=2714
2023-08-15 15:28:06.329 DEBUG (MainThread) [zigpy.zcl] [0x8C4E:7:0x0403] Received ZCL frame: b'\x18\xff\n\x00\x00)\xe8\x03'
2023-08-15 15:28:06.330 DEBUG (MainThread) [zigpy.zcl] [0x8C4E:7:0x0403] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=0, direction=<Direction.Client_to_Server: 1>, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True), tsn=255, command_id=10, *direction=<Direction.Client_to_Server: 1>)
2023-08-15 15:28:06.331 DEBUG (MainThread) [zigpy.zcl] [0x8C4E:7:0x0403] Decoded ZCL frame: PressureMeasurement:Report_Attributes(attribute_reports=[Attribute(attrid=0x0000, value=TypeValue(type=int16s, value=1000))])
2023-08-15 15:28:06.331 DEBUG (MainThread) [zigpy.zcl] [0x8C4E:7:0x0403] Received command 0x0A (TSN 255): Report_Attributes(attribute_reports=[Attribute(attrid=0x0000, value=TypeValue(type=int16s, value=1000))])
2023-08-15 15:28:06.332 DEBUG (MainThread) [zigpy.zcl] [0x8C4E:7:0x0403] Attribute report received: measured_value=1000

```

Any idea is welcome

ptvoinfo, 2023-09-12:
@glcos I've verified my BMP280. I can get other values: 1002 and 1004 in my case.

glcos, 2023-09-12:
I eventually found out that the BMP280 sensor I was using on the test device was faulty. I ordered other BMP280s and I will check again when they arrive.

closed 2023-09-27 by ptvoinfo