ptvoinfo/zigbee-configurable-firmware #223: State always restored after power off/on

obrain17, 2023-08-15:
I have two devices and in both cases **did not check _Remember state_**.
1. with CC2530 and Firmware 1.9.0.3:
[image]
2. with CC2652P and Firmware 1.9.0.6:
[image]
However, after switching power off - and on again some time later: both state and brightness appear to resume the values they had before. 
This is the behaviour I would have expected only when checking _Remember state_  (?)

ptvoinfo, 2023-08-15:
The device restores the remembered state immediately after a reboot. If it appears some time later, I think, the device connected to a coordinator and the coordinator sends a command. You can try the same with an unpaired device.

obrain17, 2023-08-15:
I have tried so:

1) Deleted (=unpaired) the CC2652P device from the coordinator
```Warning 2023-08-15 11:56:06 Device '0x00124b0022811b77' left the network```
Status Led starts blinking then = wants to join but **join is not permitted**
Now I unplug the power and wait 10 minutes, so all capacitors from the power source should be empty

2) After 10 minutes I re-plug again
Status Led still starts blinking  = wants to join but **join is still not permitted**
But **brightness of both PWM is restored - even there is evidently no connection to the coordinator**
(I can see this by the Leds on the device.)

3) I set **Permit join** on the coordinator
Device joins, interview is done, device gets a new network address etc.
(I can see this on the Wireshark sniffer as well.)
The newly joined device regularly reports the values (= restored in 2) to the coordinator

ptvoinfo, 2023-08-15:
Could you please show me all the settings of your PWM output? 
How do you check the restored state? Oscilloscope?

obrain17, 2023-08-15:
Settings are:

```
Board type: CC2652P/CC1352P2
Device type: Router
 Transmit enable (TXEN): P06, Enable high PA: P05
Serial bootloader (SBL): P15
Manufacturer name: Rainer
Model ID: ZigLight
Update firmware's timestamp : 2023-08-09
Status LED: P07, Sending reports
Set default reporting interval (s): 60

Output pins:
P11: Output 1, PWM (Hardware, 128 kHz), External pull-up (Frequency: 1 kHz, Transition time (ms): 0, Linked input: On/Off, Minimum: 0, Maximum: 255, Shift: 0, Role: Generic)
P27: Output 2, PWM (Hardware, 128 kHz), External pull-up (Frequency: 1 kHz, Transition time (ms): 0, Linked input: On/Off, Minimum: 0, Maximum: 255, Shift: 0, Role: Generic)
P40: Output 3, Source voltage
P41: Output 4, Internal temperature
P42: Output 5, Group switch (Outputs: 6, Mode: Switch)
P26: Output 6, GPIO, External pull-up (Role: Generic)

Input pins:
P15: Input 1, GPIO, Pull-up, Link to out 1, Long, Double, Tripple (Bind command: On/Off)
P14: Input 2, GPIO, Pull-up, Long, Double, Tripple (Bind command: Changes level Up)

``` 
I roughly checked the restored state visibly by the brightness of the attached LEDs. But now I made an oscilloscpe screenshot shortly after replugging the power:   (Attention: The **signal is inverted** because the oscilloscope probes are attached after the driver transistors for the LEDs.)

Yellow is Channel1 (Value 64) . Blue is Channel 2 (Value 128)
 
[image]

Before unplugging the state was shown in Zigbee2MQTT as follows:

[image]

closed 2023-10-31 by ptvoinfo