ptvoinfo/zigbee-configurable-firmware #224: MHZ19 no connection

keynote11, 2023-08-19:
Hello,

i have no connection to mhz19c with this setting from the modkam.ru_airsense.ini:

Firmware settings:
Board type: CC2530
Device type: End device without routing
Model ID: ptvo_airsense
Status LED: P01, Joining or errors
Set default reporting interval (s): 60

Output pins:
P02: Output 2, MHZ 19 (Address (dec): 1)
P04: Output 3, GPIO, Pull-up
P05: Output 4, GPIO, Pull-up

Input pins:
P20: Input 1, External wake-up, Pull-up

If i take the firmware from diyruz - Airsense i have connection to mhz19c

keynote11, 2023-09-04:
I checked today with a senseair s8 and I don't get any values here either.
With the diyruz firmware I get values from the Senseair S8.
What am I doing wrong?

I tested this on the following hardware:

CC2530
E18 MS1 PCB
E182-MS1-PA2
CC2652

Z2M:
Zigbee2MQTT-Version
1.33.0
Coordinator-Typ
zStack3x0
Coordinator-Version
20230507
Coordinator IEEE Adresse
0x00124b0022811d05
Frontend-Version
0.6.135

I save the Custom Converter ptvo_airsense on z2m and set the setting
[image]

keynote11, 2023-09-04:
i test it with firmware 1.4.11.1 - no problem, i have connection to mhz19

ptvoinfo, 2023-09-12:
@keynote11, Sorry, about the delay with my answer. Could you please try to connect UART from the module to a computer and check that the device sends a request to UART?

keynote11, 2023-09-12:
@ptvoinfo no problem.
I followed your instructions - https://ptvo.info/zigbee-configurable-firmware-features/uart/
and that worked with E18-MS1PA2 USB Testboard.

my configuration:
[image]
[image]

serialmonitor
[image]

what else can i test?

Thank you for your time!

ptvoinfo, 2023-09-27:
@keynote11 Could you please test the latest version of the firmware where this problem was fixed?

keynote11, 2023-09-29:
@ptvoinfo it works fine. Thank you!

closed 2023-09-29 by keynote11