ptvoinfo/zigbee-configurable-firmware #225: BMP280 on v 1.9 not working

olekdata, 2023-08-29:
version 1.9.0.4

txt file
```
Board type: CC2530
Device type: Router
Update firmware's timestamp : 2023-08-29
Set default reporting interval (s): 60

Output pins:
P31: Output 1, GPIO (Role: Generic), Remember state
P06: Output 2, Temperature sensor (I2C), Pull-up (Address (dec): 119, Type: BMx280)
P07: Output 3, I2C SCL/SPI CLK, Pull-up

Input pins:
P12: Input 1, GPIO, Pull-up, Switch, Link to out 1 (Bind command: On/Off)
```
screen
[image]
not working

the same on version 1.4.11.1

txt file
```
Board type: CC2530
Device type: Router
Update firmware's timestamp : 2023-08-29
Set default reporting interval (s): 60

Output pins:
P31: Output 1, GPIO, Remember state
P06: Output 2, BMP280 (I2C SDA), Pull-up (Address (dec): 119)
P07: Output 3, I2C SCL, Pull-up

Input pins:
P12: Input 1, GPIO, Pull-up, Switch, Link to out 1
```
screen
[image]
working normal

maybe i am doing something wrong

Regards
Alexander

ptvoinfo, 2023-08-29:
@olekdata Hi, 
Where do you watch incoming data? If it is Z2M, do you use a custom converter? What version of Z2M do you use? Did you try to enable debug logs in Z2M?

olekdata, 2023-08-29:
Version Zigbee2MQTT : 1.28.0-dev commit: [ddc4345](https://github.com/Koenkk/zigbee2mqtt/commit/ddc4345)

get state with CC2530 on version 1.4 
```
{
    "device_l2": "77",
    "humidity_l2": 60.5,
    "l2": 60.453,
    "linkquality": 149,
    "pressure_l2": 98976,
    "state_l1": "OFF",
    "temperature_l2": 22.4,
    "action": null,
    "l1": null,
    "l10": null,
    "l11": null,
    "l12": null,
    "l13": null,
    "l14": null,
    "l15": null,
    "l16": null,
    "l3": null,
    "l4": null,
    "l5": null,
    "l6": null,
    "l7": null,
    "l8": null,
    "l9": null,
    "state_l10": null,
    "state_l11": null,
    "state_l12": null,
    "state_l13": null,
    "state_l14": null,
    "state_l15": null,
    "state_l16": null,
    "state_l2": null,
    "state_l3": null,
    "state_l4": null,
    "state_l5": null,
    "state_l6": null,
    "state_l7": null,
    "state_l8": null,
    "state_l9": null
}
```

with varsion 1.9 in state not value : humidity_l2, pressure_l2, temperature_l2 etc

glcos, 2023-08-29:
@olekdata I have the same issue with BMP280, see here https://github.com/ptvoinfo/zigbee-configurable-firmware/issues/222#issuecomment-1678949396

@ptvoinfo in the new firmware release, please keep the same precision, that is 2 digits after the comma, that was in the previous versions. This is very useful in weather monitoring applications.

olekdata, 2023-08-29:
Problem on Z2M side?
Can i change something set in Z2M?

ptvoinfo, 2023-08-29:
@olekdata You need the latest version of z2m or a custom converter for your device because the new firmware sends temperature in the other format that the old version of z2m expects.

ptvoinfo, 2023-09-12:
@glcos, I've double-checked. The firmware sends a temperature with 2 digits after the decimal point.

glcos, 2023-09-12:
Thanks for your feedback, I will perform other tests as soon as I can.

closed 2024-03-31 by ptvoinfo