ptvoinfo/zigbee-configurable-firmware #226: Model ID won't update

phalox, 2023-09-02:
Referring to a closed issue #178

I seem to be experiencing the same issue:

[image]

[image]

I've deleted the device in Z2M, and restarted Z2M. None of it seems to fix the problem. When I make changes to the firmware (e.g. add an input), they do show up in Z2M.

I've also let it validate that the hex file is correct, and that seems to be the case:

[image]

I bumped into one more thing.... When exporting the converter and opening the file, I still see the old name (ptvo.switch).

[image]

Could it be that this is part of the issue?

Also, using the custom converter in Z2M, I think it's still not triggered because of the lack of another modelID

_Originally posted by @phalox in https://github.com/ptvoinfo/zigbee-configurable-firmware/issues/178#issuecomment-1703841926_

ptvoinfo, 2023-09-04:
@phalox Hi,

1. Delete device in Z2M.
2. Stop Z2M.
3. Delete coordinator_backup.json in the "data" folder.
4. Start Z2M.
5. Join the device again.

phalox, 2023-09-04:
Great, that seems to have solved it! 

[image]

What would be your recommended development process? When playing around with different settings, adding & removing sensors etc, this seems a little tedious. I also found it challenging to debug zigbee issues. Often it's a matter of reflashing/repowering, but I'd love some more visibility.

closed 2023-09-04 by phalox

ptvoinfo, 2023-09-04:
Unfortunately, Z2M is not friendly for DIY :). I use the same procedure while testing/