ptvoinfo/zigbee-configurable-firmware #227: Arduino and UART

andrejsoucek, 2023-09-04:
Hello, sorry for posting a question to issues, however I can't figure out following problem with Arduino and CC2530+CC2591.
In PTVO, I set Output 1 to P02, UART, 9600 baud, Packet end None, Byte 239.
I connected CC2530 P02 to Arduino pin 5, P03 to pin 6 and then use SoftwareSerial library: 
```
SoftwareSerial ss = SoftwareSerial(6, 5) // rx, tx;
ss.begin(9600);
```
The CC2530 is connected to my network, recognized by zigbee2mqtt, I see availability topic in MQTT Explorer, however I can't transfer any data in any direction.
I tried ss.println('1'); and nothing happens in mqtt explorer, other way I tried to publish
```
Topic: zigbee2mqtt/[friedly_name]/set
Payload: {"action": [2, 0, 0]}
```
with no results.

What am I doing wrong? Am I missing something?

andrejsoucek, 2023-09-04:
Ok I have some progress, I've discovered debug logs in Z2M saying that it receives messages but they didn't get processed because the device joined as a CC2530 router. I stopped Z2M, deleted the backup json, started again, now the device looks like a DIY device according to PTVO Expert settings, I imported the external converter, everything looks ok, the device is "Supported" in Z2M, exposes linkquality and action. But there are no logs of receiving any messages now.

andrejsoucek, 2023-09-04:
Ok, looks like I was sending some unsupported characters. Changed the message to be a json: `{"1": 0}` which is now visible in logs. 
```
Info 2023-09-05 01:10:41MQTT publish: topic 'zigbee2mqtt/[name]', payload '{"action":"{\"1\": 0}","linkquality":148}'
Info 2023-09-05 01:10:41MQTT publish: topic 'zigbee2mqtt/[name]', payload '{"action":"","linkquality":148}'
Info 2023-09-05 01:10:41MQTT publish: topic 'zigbee2mqtt/[name]/action', payload '{"1": 0}'
```
But why it gets reset to empty immediately after it is received? How do I use the value in HA?

ptvoinfo, 2023-09-06:
@andrejsoucek 
It reset because a new data packet may appear immediately after it, but it may have the same content.

andrejsoucek, 2023-09-07:
> @andrejsoucek It reset because a new data packet may appear immediately after it, but it may have the same content.

how do I prevent it? or where would that packet come from?

ptvoinfo, 2023-09-15:
@andrejsoucek You can create a custom converter for Z2M and process data from a device as you want.

andrejsoucek, 2023-09-19:
I ended up with this converter for now: https://github.com/andrejsoucek/zigbee-weather-station/blob/master/soucek-weather.js
which fixed merged payloads (2 serial prints actually appeared as one payload, see below) but don't know how to skip the empty action

Arduino code:
```
Serial.print("{\"1\":" + value1 + "}");
Serial.print("{\"2\":" + value2 + "}");
```

PTVO generated converter payload:
`{ 1: "value1" }{ 2: "value2" }`

Edited converter payload:
`{ 1: "value1", 2: "value2" }`

closed 2023-09-19 by andrejsoucek