ptvoinfo/zigbee-configurable-firmware #228: Sonoff Zigbee USB Dongle Plus P - freeze with preset Setting

keynote11, 2023-09-05:
I want to flash my Sonoff ZigBee Stick with Ptvo. I take the sonoff_3_usb_dongle_plus file and adjust it using the firmware configuration program. I set the signal strength to 20.

z2m found the device, but the device is unreachable after 2 minutes.

z2m
Zigbee2MQTT version
1.33.0 commit: f224ef7b
Coordinator type
zStack3x0
Coordinator version
20230507
Coordinator IEEE address
0x00124b0022811d05
frontend version
0.6.135

ptvoinfo, 2023-09-06:
@keynote11 Why do you think that this problem is related to the firmware?

keynote11, 2023-09-06:
@ptvoinfo what am I doing wrong?

my setting:
[image]

[image]

i test this on a other board https://github.com/knowic/cc2652p-dongle/blob/main/cc2652p-dongle-1v2.pdf
without the outputs and here the same. The board is freeze after 2 min.

ptvoinfo, 2023-09-07:
What do you mean by "unreachable"?
Does the status led show anything when you press a button on the stick?

keynote11, 2023-09-07:
@ptvoinfo no the status led is always on.
i mean with unreachable, the device is not available. 
If i reboot the device over reset power or reset button, on z2m stands on last seen "just now" and then the times run. After 1 min comes again "just now" but after this the times run and run and run. If i send over the dev console a command i receive a "AREQ - ZDO - bindRsp after 10000ms". 

if i set the default reporting interval the device always comes online after this time?

ptvoinfo, 2023-09-07:
The status LED should blink when the device joins or unable to send data. You can also change the status led mode to check that the device is alive.

keynote11, 2023-09-11:
@ptvoinfo i test it with this board https://github.com/knowic/cc2652p-dongle/blob/main/cc2652p-dongle-1v2.pdf
is the setting "Board type" = CC2652p/CC1352P2 right?
If so, I tested it with a GPIO input switch and led state with sending reports. The switch works and I see it on z2m. But 2 minutes later nothing happens on z2m. It stops responding. When I switch, the LED is on.
it´s a bug on z2m oder firmware?

ptvoinfo, 2023-09-11:
@keynote11 
It looks like this device uses pin P05/P06 to control a power amplifier (select it on the Expert tab).

keynote11, 2023-09-11:
@ptvoinfo if I select pin 5 & 6 the red LED turns on.
I think pin 29 & 28 is correct.
[image]

keynote11, 2023-09-13:
@ptvoinfo Today I tested the CC2652P board with the same UART connection as yesterday with the CC2530. 
I can send and receive data in the first two minutes. After that no more reaction.
Could you please check this with your device?

ptvoinfo, 2023-09-14:
@keynote11 I do not see UART in your firmware configuration. What is a role of this?

keynote11, 2023-09-14:
UART doesn't play a role at first. Maybe I would like to connect a Senseair S8 later. 
I just wanted to test the communication after 2 minutes. 

what do you think its a bug in the firmware or z2m?
can I test something else?

ptvoinfo, 2023-09-14:
Many users of PTVO use the same hardware/firmware, but I didn't receive similar bug reports before (last time).

Did you check this device on a separate Zigbee network? Probably, it is incompatible with other devices in your existing network.

keynote11, 2023-09-14:
@ptvoinfo many thanks for this tip!
Both boards (routers) work in a separate Zigbee network with Sonoff Stick as coordinator.
Do you have any advice or experience on how I can find this device that is causing problems? Or is the Coordinator the problem?

Zigbee network:

Coordinator: ZigStar Lan Gateway

Total 53
By device type
Router: 32
End devices: 17
GreenPower: 4

By power source
Mains (single phase): 32
Battery: 17
undefined: 4

By vendor
LUMI: 11
innr: 7
_TZ3000_gjnozsaz: 6
GLEDOPTO: 4
Signify Netherlands B.V.: 4
undefined: 4
_TZ3000_typdpbpg: 3
ptvo.info: 3
3A Smart Home DE: 2
_TZ3210_dbilpfqk: 1
Philips: 1
_TZ3210_dwytrmda: 1
IKEA of Sweden: 1
_TZ3000_ltt60asa: 1
_TZ3000_fllyghyj: 1
_TZ3000_cfnprab5: 1
TexasInstruments: 1
_TZ3000_tqlv4ug4: 1

By model
TS011F: 10
lumi.weather: 5
ROM001: 5
RB 279 T: 4
GreenPower_254: 4
lumi.remote.b1acn01: 3
GL-C-006: 2
RB 278 T: 2
LXN59-1S7LX1.0: 2
GL-C-007: 1
TS0501B: 1
GL-C-008P: 1
TS130F: 1
TRADFRI bulb E27 WS opal 980lm: 1
TS0004: 1
lumi.sensor_ht: 1
TS0201: 1
lumi.sen_ill.agl01: 1
lumi.sensor_magnet.aq2: 1
RB 178 T: 1
ptvo.switch: 1
ti.router: 1
TS0001: 1
ti2.router: 1
ptvo.air: 1

ptvoinfo, 2023-09-15:
I've encountered with the similar problem one time when one of tuya wall switches started reconnecting to a network every few seconds, and it made routers unresponsive.
Diagnosing: I've used CC2531 as a sniffer and monitored network traffic.
Resolving: 
1. Try to reset and rejoin the broken device.
2. In my case, only the rebuilding of the whole network helped (I've changed the network key and pan ID).

keynote11, 2023-10-02:
@ptvoinfo I tested it again in a separate zigbee network. In this there is only the coordinator and the cc2652p device. After 20 minutes the device is offline.
it is very strange…