ptvoinfo/zigbee-configurable-firmware #229: HX711 and Home Assistant

epvdvalk, 2023-09-08:
Hi, Thanks for the great software. I have made a scale using load cells and a HX711 amplifier. No issues with programming the CC2652 module and the integration is also ok. Saved a custom converter, also ok. Th thing is the calibration. My sclae factor switch is not available in HA/Z2M. What am I doing wrong here?  See images:
[image]
[image]

ptvoinfo, 2023-09-08:
@epvdvalk Hi, I see the scale factor field in your Z2M. What is wrong, exactly?

epvdvalk, 2023-09-08:
Hi ,thanks for the fast reply!

The switch for “enable zero calibration” is not available in Z2M.

Kindest regards, Epko.

epvdvalk, 2023-09-08:
It also says “no converter available for state” (undefined)

epvdvalk, 2023-09-22:
Hi, I finally got it working and calibrated the scale. I use the scale to measure the amount of salt in my water softener. Measured the salt weight and calibrated the scale. It started at 20 kg's but after a day it started drifting and after a week the scale shows -37kgs. How is this possible? Is there a fix for it?
[image]

ptvoinfo, 2023-10-31:
@epvdvalk Sorry, but I cannot repeat this problem in one-day test. Can something enable/disable the tare mode automatically?

epvdvalk, 2023-11-03:
Hi, It is just not working as expected unfortunately. I have switched to a different type of load cell (the type you are using) but still there is no switch that enables the tare function unless I program a switch in the firmware.

epvdvalk, 2023-11-03:
Can I add multiple HX711 modules to 1 CC2652P module?