ptvoinfo/zigbee-configurable-firmware #230: Addition Radar sensor LP 2410

j-Pascal, 2023-09-10:
Hello, will you be able to add the LD 2410 sensor for the realization of radar for static people.

Thank you for your work.

https://www.hlktech.net/index.php?id=988

Wielebny666, 2023-09-25:
It would even be useful in the premium version.

ptvoinfo, 2023-09-27:
As I see, this sensor has an output with HIGH (occupied) and LOW (unoccupied). So, you can connect it to PTVO as a regular presence sensor:
https://ptvo.info/zigbee-configurable-firmware-features/diy-examples/diy-zigbee-motion-detection-sensor-pir/

Wielebny666, 2023-09-29:
I agree, but it is not possible to change LD2410 parameters such as sensitivity, range, etc :-(

j-Pascal, 2023-09-29:
Indeed, the great interest is to be able to read the movement distance, static as well as the detection distance.

ptvoinfo, 2023-09-29:
@Wielebny666 This sensors requires a proprietary program to change sensor settings via UART.

Wielebny666, 2023-09-30:
I know, but when the sensor is mounted in a case, e.g. under the ceiling, this is an inconvenient solution. 
The ESPHome project includes full support for this sensor via UART.