ptvoinfo/zigbee-configurable-firmware #233: Analog values with new UART sensor

obrain17, 2023-09-17:
Floating point values are not transmitted correctly over the UART.

1. Send this message to the UART:

[image]

2. Correctly logged in Zigbee2Mqtt:

[image]

3. I would expect the value 3DE147AE being sent to the UART:

[image]

4. Instead AEAEAEAE is sent:

[image]

Also for other values always the last byte is repeated 4 times.

ptvoinfo, 2023-09-18:
@obrain17 I've fixed both bugs related to the UART sensor.

obrain17, 2023-09-18:
Thank you for the quick reaction.

I just checked both `Analog value` and `Level Control` in both directions and all values are transmitted correctly.

So I am closing the issue now.

closed 2023-09-18 by obrain17