ptvoinfo/zigbee-configurable-firmware #235: Bind command executed after Trigger

obrain17, 2023-09-18:
I have a modified ZBMini with this configuration:

[image]

Input 2 is linked to Output 2 and has the Bind command "Toggle". For this one it is bound to a device called AmpPlug:

[image]

This "Toggle" command is executed whenever the Switch on **Input 2** is switched - as expected.

Now I found that the Bind command is also executed when **Output 2** is set to "off" after a "Trigger" command (`'onWithTimedOff'`) has timed out.  

I think state changes on Outputs should not affect Bind commands on Inputs that are linked to them. Normal "on" and "off" and starting the  "onWithTimedOff" on do already correctly not affect the Bind command.

Behaviour is for both CC2530 and CC2652.

ptvoinfo, 2023-09-25:
@obrain17 This output is linked to an input.  So, yes, it can affect the input.

obrain17, 2023-09-25:
I checked with a different setup, with Input not linked to the Output:

[image]

If the Input 2 is not linked to the Output 2, the behaviour is still the same:
The Bind command "Toggle" is sent when Output 2 is set to "off" after a "Trigger" command ('onWithTimedOff') has timed out. 

Probably the internal timer callback function additionally executes all binding commands for endpoint 2 rather than just switching the Output off and report this.