ptvoinfo/zigbee-configurable-firmware #236: UART TX CC2652P tidak mengirim serial

t0p1x, 2023-09-23:
I configured zigbee CC2652P router mode end device mode does not send any serial

For coordinator I use Tasmota
serial log results when connected to an end device like this

15:08:06.310 ZIG: ZbZNPSent 250428DA28DA01
15:08:06.360 ZIG: {"ZbZNPReceived":"650400"}
15:08:06.411 ZIG: {"ZbZNPReceived":"458428DA0028DA14010401FEFF000400001400070006000200000600"}
15:08:06.416 RSL: RESULT = {"ZbState":{"Status":33,"Device":"0xDA28","Endpoint":"0x01","ProfileId":"0x0104","DeviceId":" 0xFFFE","DeviceVersion":0,"InClusters":["0x0000","0x0014","0x0007","0x0006"],"OutClusters":["0x0000","0x0006"]}}

ptvoinfo, 2023-09-25:
@t0p1x Could you please describe how do you test UART on CC2652? Did you try to swap TX and RX wires?

t0p1x, 2023-09-25:
data sent from the end device to the coordinator was successful, but not from the coordinator to the end device
for endpoints 1 and 2, the GenMultiStateValue part is input only
[image]

t0p1x, 2023-09-25:
Thank you for your help
Finally I can send serial data using zigbee2mqtt
In the tasmota program there is no command to send serial data

closed 2023-09-27 by ptvoinfo