ptvoinfo/zigbee-configurable-firmware #237: Two DS18B20, only one working

Enjoy-Combi, 2023-09-29:
Hi,

First try with ptvo.
My project consists in 2 DS18B20 and one relay, here after my config:
[image]

Relay works good. Only one temperature is reported. I was thinking first it was due to a faulty sensor. I have unsoldered one of them, the remaining one works. I did the same with the second one (only this one soldered), and the second one works as well. They both works when alone in the circuit. When I have the two ones, only one works.

Any idea what is wrong ?
[image]

One more question, I don't figure out the logical between input & output.  For me the two ds18b20 behave as an input and the gpio behaves as an output. And they are all selected as output in my config (as per some examples I have seen on the net).

Many thanks for your help !

ptvoinfo, 2023-09-29:
@Enjoy-Combi 
1. You should specify your sensor ID in the corresponding field.
2. Input/Output role is important for GPIO. It does not affect other sensors.

Enjoy-Combi, 2023-09-30:
1. Noted, I tried to do it and then no temperature at all. I mean, no field temperature anymore, only the one for the switch. Will try to do it again. I assume there is no order to respect, anyway I will try in both order.
2.Thank you for that information !

ptvoinfo, 2023-09-30:
1. Try to rejoin the sensor...

Enjoy-Combi, 2023-10-01:
What I get with this config:

[image]

I removed the sensor and pair it again:

[image]

Enjoy-Combi, 2023-10-01:
I changed the order of the temp id:
[image]

Removed & rejoined the sensor and same issue. I am lost

ptvoinfo, 2023-10-02:
Do you have the external 4.7k resistor on P15?

Enjoy-Combi, 2023-10-02:
Yes I have.

Crazy point, I went to the log and I see:
2023-10-02 09:03:59MQTT publish: topic 'zigbee2mqtt/0x00124b000959024a', payload '{"linkquality":72,"state_l3":"OFF","temperature_l1":23.06,"temperature_l2":22.93}'

What is OK.

And the entities are well available:

[image]

Only surprising that all entities are not available into Zigbee2mqtt.
I fear that I didn't fully follow the tutorial about DS18B20, have to create a sensor to gather all entities.

Then fixed, many thanks for your support

closed 2023-10-02 by Enjoy-Combi